eez-open/studio #677: [LVGL] Focus changes occur for all screens and widgets, not just the active one

RamasyaR, 2024-12-24:
**Describe the bug**
Focus changes occur for all screens and widgets, not just the active one.

**Reproduce**
Steps to reproduce the behavior:
1. Run the application (Linux amd64)
2. Rotate the mouse wheel
3. After two mouse rotations, the focus will be set to some widget out of the visible area
4. The more widgets you can focus, the more mouse rotations you need to make to return to the original position

**Expected behavior**
Focus should only switch between objects on the active screen

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Debian
 - Version: 12

**LVGL version**
- 9.x

**Additional context**
Test application: [esp_lvgl_ui_x11_demo.zip](https://github.com/user-attachments/files/18238489/esp_lvgl_ui_x11_demo.zip)
Source code of the test application: [test_focus.zip](https://github.com/user-attachments/files/18238462/test_focus.zip)

mvladic, 2024-12-24:
This is normal behavior of the LVGL when `lv_group_set_default` is called, i.e. it will add every widget created to the group no matter if widget is visible or not.

RamasyaR, 2024-12-24:
Thanks for the answer. Since this is a standard behavior for LVGL, it is possible to create a group for each screen and add all widgets belonging to this screen to this group. Because it is unlikely that anyone will need this behavior by default

RamasyaR, 2024-12-24:
What is the workaround for now? Create a group for each screen yourself and add widgets there? And probably change the label to "improvement".

mvladic, 2024-12-24:
Studio has support for the group. You can find more details here: https://github.com/eez-open/studio/issues/229

There is no need to create a group per screen. You can create a single group per input device, for example encoder (simulated with the mouse wheel on Desktop), and then decide which widgets on particular screen you want to be in that group. Studio will take care on screen transitions to remove widgets from previous screen and add widgets of the current screen to the group.

RamasyaR, 2024-12-24:
Yes, it works, we can close the problem, thanks

closed 2024-12-24 by mvladic