eez-open/studio #681: [LVGL] Led Color uses incomplete `lv_led_t` typedef

arturv2000, 2024-12-29:
**Describe the bug**
Setting the LED color to be setup via a vatiable, the code generated cannot compile

```c
{
        int32_t new_val = get_var_cable_status_color();
        uint32_t cur_val = lv_color_to_u32(((lv_led_t *)objects.led_cable_status)->color);
        if (new_val != cur_val) {
            tick_value_change_obj = objects.led_cable_status;
            lv_led_set_color(objects.led_cable_status, lv_color_hex(new_val));
            tick_value_change_obj = NULL;
        }
    }
```

Error generated:

```txt
../Middleware/eez-gui/Feira_GUI/src/ui/screens.c:355:85: error: invalid use of incomplete typedef 'lv_led_t'
  355 |         uint32_t cur_val = lv_color_to_u32(((lv_led_t *)objects.led_actuator_status)->color);
      |                                                                                     ^~
../Middleware/eez-gui/Feira_GUI/src/ui/screens.c:364:82: error: invalid use of incomplete typedef 'lv_led_t'
```

**To Reproduce**
Steps to reproduce the behavior:
1. Add a LED to a screen
2. Create a variable for the LED Color (in my case of type int32_t)
3. Assign the LED Color to the variable
4. See error

**Expected behavior**
No error being generated by compiler

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [Windows 11]
 - Version [0.21.0]
- Compiler: gcc version 10.3.1 20210824 (release) (GNU Arm Embedded Toolchain 10.3-2021.10)

** LVGL version (if used)**

- 9.x

**Additional context**
Add any other context about the problem here.

mvladic, 2024-12-30:
Thanks for the detailed analysis of the problem. I am aware of this problem from before and the LVGL project that uses EEZ Flow does indeed include this private header file, but I forgot to do the same in the project that does not use EEZ Flow.

As a workaround for now you can add this include in `screens.c` template file in Settings:

[image]

erik-toth, 2026-09-29:
Hi, I also bumped into the same problem... the bug seems to be that the code generated tries to use the incorrect API to interface the led color. Using the private API is not recommended. A proper fix would be to change the code generation to the right public API as provided by the lvgl library.

The problem is this specific line:
```cpp
uint32_t cur_val = lv_color_to_u32(((lv_led_t *)objects.led_widget)->color);
```

To fix it, we should use:
```cpp
uint32_t cur_val = lv_color_to_u32(lv_led_get_color(objects.led_widget))
```

What's interessting is that further down, that the correct set-methode is used....

So the whole corrected block should be:
```cpp
{
    uint32_t new_val = get_var_led_preview();
    new_val = lv_color_to_u32(lv_color_hex(new_val));
    uint32_t cur_val = lv_color_to_u32(lv_led_get_color(objects.led_preview_icon));
    if (new_val != cur_val) {
        tick_value_change_obj = objects.led_preview_icon;
        lv_led_set_color(objects.led_preview_icon, lv_color_hex(new_val));
        tick_value_change_obj = NULL;
    }
}
```

Kind regards,
Erik