eez-open/studio #687: background color does not work for most components

AlexanderGrathIBO, 2025-01-08:
**Describe the bug**
For some components (I can confirm label, image and container) the background color does not appear.

**To Reproduce**
Steps to reproduce the behavior:
1. create a LVGL project (8.x)
2. add a label to the screen
3. change background color
4. nothing happens, neither in the preview, nor when you click run, nor when I upload it to my device/screen

**Expected behavior**
For the background color to change.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [macOS]
 - Version [15.1.1]
 - Apple M2

** LVGL version (if used)**

- 8.x

**Additional context**
It will change the code of screens.c, button works without problems and the code looks basically identical.
Potentially related issue: #644

RamasyaR, 2025-01-08:
Change the opacity to 255 and you will see the background

AlexanderGrathIBO, 2025-01-09:
So for anybody finding this with similar problems:
First check the opacity, some components have it set to 255 as the default value, some don't. You can set a new style, change it to the default value you want and than use that as the new default for the component.

The issue I had and why I opened this issue:
If you have a **smaller screen** make sure you can actually see all the UI elements. There is a button at the top right (within every sub-window) to enlarge it. 
I thought I was going insane, sometimes things would work sometimes not, turns out I **couldn't see all the settings** and I was changing things in 'selected part'/'Ticks part' etc. and was wondering why none of the changes would do anything to the default behavior.

mvladic, 2025-01-09:
We are using LVGL default values for the opacity. For example, Button widget has 255 opacity by default and Label widget has 0.

mvladic, 2025-01-09:
> I would suggest setting that to 255 on default (when color is set).

This makes sense. We can change opacity to 255 if it was 0 when color is changed.