eez-open/studio #688: Add RGB565 support

RogerRU, 2025-01-09:
**Is your feature request related to a problem? Please describe.**
There are many  ESP32 Cheap Yellow Display now. All of them support only RGB565 colors. EEZ Studio supports only RGB888.
In such a situation, it is very difficult to create a design because the colors in the editor do not match the colors on the display.

**Describe the solution you'd like**
Add RGB565 colors support to EEZ Studio

Thx

zalexzperez, 2025-01-09:
Studio already supports RGB565 and RGB565A8:

[image]

RogerRU, 2025-01-10:
> Studio already supports RGB565 and RGB565A8:
> image.png (view on web)

No.
 
As far as I understand, what you are writing about applies only to images, or more precisely to color coding in the generated file.
 
If you edit the color of widget elements, the color selection dialog will give you a color in 24-bit format (RGB888). Thus, the generated file will have a color like 0xFFXXXXXX, so if the display does not support 24-bit color, but only 16 bits, there will be strong color distortions.
 
IMHO this is a rather serious issue, since developing a design for 16-bit colors displays is either impossible or very difficult.
PS
 
Competitors have solved this problem, I am talking about SquareLine Studio
 
Thx

RogerRU, 2025-01-12:
Hi all again))
I've been working on my problem for a while and this is what I've found out so far:
1. Studio works fine, the problem is not in it.
2. The problem is not in my monitor profiles

Since the problem is not in Studio, you can close the thread, but I still ask you to help me with this case.
I'm attaching screenshots of what I see in Studio and what's on the display screen.

As you can see, the problem only occurs on certain color ranges in equal intervals. (the panel is filled with a gradient from 0xFFFFFF to 0x000000) On gradients of other colors, the problem is also present, but less noticeable. The gradient itself has nothing to do with the problem, since with a normal fill from the specified range there is also a distortion.

**What I've already done:**
1. Played with color support in LVGL (RGB565, RGB565A8, RGB888, ...)
2. Checked the display driver settings and parameters 10 times and compared them with the settings of other users.
**My guesses:**
1. I did something wrong in the LVGL settings
2. Incorrect conversion by LVGL from 24 bit to GRB565 because the stripes are more yellow or green, then maybe something is wrong with the Green bits because there are 6 of them
3. My display is damaged or defective

Studio view:
[image]
[image]

Display view:
[image]
[image]

closed 2025-02-04 by mvladic