eez-open/studio #695: Allow making flow lines straight when dragging widget

zalexzperez, 2025-01-12:
I like my flow lines straight but this gives me a little OCD 🥹  :
[image]

https://github.com/user-attachments/assets/df6046ab-5346-4a40-a385-aa32c37619ec

The auto snapping feature does not always accomplish straight line due to differences in widget heights.

I'm aware holding the SHIFT key activates the free-dragging mode by disabling auto snapping, but even then, it's not easy to get lines completely straight.

Would it be possible to also add straight-line matching to the auto-snapping feature? 
If not possible, maybe add it another key press, like CTRL.

mvladic, 2025-01-13:
You can do that with Up and Down keyboard keys which moves selected action (or widget) by 1 pixel:

[image]

mvladic, 2025-01-14:
I added CTRL key shortcut to automatically align connection line.

zalexzperez, 2025-01-14:
Thank you, works great!!

zalexzperez, 2025-01-17:
I noticed the first widget in a flow branch doesn't have any previous widgets to use as a reference for constraining the line, so the new feature doesn't work on the very first widget.
Please, add support for this missing case.

mvladic, 2025-01-20:
Fixed.

I'm now selecting closest connection line for both cases, i.e. when dragged component is either source or target component in the connection line.

closed 2026-06-24 by mvladic