eez-open/studio #700: Arduino 2.3.4 compile ui folder, No .o .a file output, link fail

kylongmu, 2025-01-14:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. copy ui directory to arduino project folder.
2. Click on 'go'
3. See error
[image]

[image]
open the sketch folder, other code has .o output
[image]
but the ui has nothing!
[image]

** LVGL version (if used)**

9.x

**error messagge**
D:\Develop\Arduino\ESP32P4\1-Demo\Demo_Arduino\1_2_Lvgl_V9\esp32p4_arduino_mipi-dsi_lvgl/esp32p4_arduino_mipi-dsi_lvgl.ino:72:(.text._Z5setupv+0xea): undefined reference to `ui_init'
C:/Users/kylon/AppData/Local/Arduino15/packages/esp32/tools/esp-rv32/2405/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\kylon\AppData\Local\arduino\sketches\F53207CEB6ADE9A5B50BAA99B7D9CC15\sketch\objs.a(esp32p4_arduino_mipi-dsi_lvgl.ino.cpp.o): in function `loop()':
D:\Develop\Arduino\ESP32P4\1-Demo\Demo_Arduino\1_2_Lvgl_V9\esp32p4_arduino_mipi-dsi_lvgl/esp32p4_arduino_mipi-dsi_lvgl.ino:67:(.text._Z4loopv+0x8): undefined reference to `ui_tick'
collect2.exe: error: ld returned 1 exit status

mvladic, 2025-01-14:
All Studio generated files must be inside the same folder where INO file is, i.e. it can't be in the `ui` subfolder. This is required by the Arduino IDE and its build system.

kylongmu, 2025-01-14:
> All Studio generated source files must be inside the same folder where INO file is, i.e. it can't be in the `ui` subfolder. This is required by the Arduino IDE and its build system.
No, this is not the reason, look at the touch folder, my other code is ok.
And I tested put all file at the same folder, this error is same.

mvladic, 2025-01-14:
If there is no `.o` file then you didn't compile the `.c` file. You should add studio generated source files (`*.c`) to your build system. Sometimes, it is not enough just to copy files to some folder, you also need to add these files to your make file.

mvladic, 2025-01-14:
> No, this is not the reason, look at the touch folder, my other code is ok.

Where is touch folder located relative to your INO file?

kylongmu, 2025-01-15:
> I just found out that recent Arduino IDE does allow subfolder, but everything must be under `src` subfolder, check this: https://forum.arduino.cc/t/subfolders-in-sketch-folder/564852
> 
> In your case, `touch` is under `src` subfolder but `ui` is not:
> 
Thanks for your help, change the ui folder under, the compile is OK!

closed 2025-01-15 by mvladic