eez-open/studio #705: LVGL 'label' widget's generated code issue

GCTechnology, 2025-01-18:
**Describe the bug**
LVGL 'label' widget's generated code does not look right while binding with an integer variable.

**To Reproduce**
Steps to reproduce the behavior:
1. Create an integer variable.
[image]

2. Create a 'label' widget.
[image]

3. Bind the variable with the Label widget.
[image]

4. Build and generate code.
5. Inside the code:
[image]

get_var_light_value_pre is 'int32' type, and const char *new_val is a 'char *' type, which is not correct.

The code should be look like:
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 10 22H2

** LVGL version (if used)**
- 8.x

pragmaDave, 2025-02-24:
I'm experiencing the same issue with variables I've created in EEZ Studio of type int and float. The function declarations generated in vars.h agrees with the data types defined for the variables in EEZ Studio. However the get_variable_ function used in screens.c always expects a char* as return type. Thus the compiler error and the need to write corrective code to correct the issue after every build. (Note that this problem did not occur when I was using Flow. I removed Flow and the issue started.)

mvladic, 2025-02-24:
Currently, Label widget only support string variables. So, change variable type to string and do conversion from integer to string in your code. 

I will close this issue since it is duplicate of #347.

pragmaDave, 2025-02-25:
I'm no expert so I apologize if my suggestion is wrong, but one way I think you can help with this issue is to build the get_variable_ function in vars.h with a return type of const char*. (Leave the set_variable function with the argument type as set in EEZ Studio.) This would prevent this from having to be manually changed after each build.

Not a big deal but maybe something for the next version. And btw, thanks very much for making EEZ Studio available! It's great!

mvladic, 2025-02-25:
Maybe, I wasn't completely clear in my previous comment. You created in EEZ Studio variabled called `light_value_pre` of **integer** type:

[image]

So, because it is integer variable studio expects following functions to be defined:

```
int32_t get_var_light_value_pre();
void set_var_light_value_pre(int32_t value);
```

But, Label widget currently can't work with integer variables, it can only work with the string variables, thus compiler error. 

If `light_value_pre` was defined as string variable, you will then need to define following functions and generated code would compile:

```
const char *get_var_light_value_pre();
void set_var_light_value_pre(const char *value);
```

[image]

If you want to display some integer number in Label widget then you can implement get function like this:

```
char light_value_pre_str[20] = { 0 };

const char *get_var_light_value_pre() {
    snprintf(light_value_pre_str, sizeof(light_value_pre_str), "%d", light_value_pre_int);
    return light_value_pre_str;
}
```

Definitely, Studio can be improved in this particular case:

- It can report an error when integer variable is selected for the Label Text.
- or, It can do automatic conversion from integer or float to string.

This is way #347 is still open as I plan to do something about this issue.