eez-open/studio #706: [LVGL] Add a way to control when screen is created

mvladic, 2025-01-19:
Currently, all the screens are created when `ui_init()` call is made. We should have an option `Create at start` for the page (enabled by default). It should be possible from both flow and code to:

- Create a screen object
- Delete a screen object
- Find out if screen object is created

zalexzperez, 2025-01-19:
This is a promising idea! For clarity's sake let's call it the **modular** approach if you will.

But let me propose a different one here: the **group** approach. It would be similar to what we already have with widgets and the encoder/keypad groups for navigation. 

With the group approach, the programmer wouldn't have to worry about managing which screen objects are already created and which ones require deleting. Upon adding new screens to the project in the future, the programmer would only have to assign the screen group to them. So it would escalate better. And to make it as flexible, a screen could belong to multiple groups.

So, when `ui_init()` call is made, the default group is loaded. Afterwards, the programmer would only use the `Change screen group` action widget/function.

What do you think? 
Assuming what I said is right, I think it seems to be considerably harder to implement anyway, so it may not be worth it.

mvladic, 2025-01-19:
> ... the group approach ...
> What do you think?

I think we should eventually address this in phase 2, because it is more complex to implement.

> How would all of this play with the Change Screen / Change to Previous Screen actions and the EEZ functions?

Change Screen / Change to Previous Screen will automatically create screen if it is not currently created. 

We can also automatically delete screen if it is not active anymore. But, this should be an option where you can choose to "Keep in memory" or "Remove from memory" when inactive.

> What if the screen we are requesting belongs to an inactive screen group?

In that case we can switch to that group - I suppose switch means: release screens of current active group and create screens of the new group and make it active.  But, I can think of the situation in the really complex application where you want multiple groups to be active at the same time.

----

I have on question for you. In your example where you can have normal execution of the app and OTA update execution (10 additional screens), does this means that either app is run in Normal mode or OTA Update mode, i.e. you are going from one mode to another only after reset?

zalexzperez, 2025-01-19:
> I have on question for you. In your example where you can have normal execution of the app and OTA update execution (10 additional screens), does this means that either app is run in Normal mode or OTA Update mode, i.e. you are going from one mode to another only after reset?

Yes, in my case the app only works in one mode at a time and the working mode is changed after a reset.

By doing this, I managed to reduce system heap usage and have a working OTA update mode at the expense of the normal mode, as the added screens for the updater part demand an increase in LV_MEM_SIZE. However, increasing that memory pool compromises the update mode again.

mvladic, 2025-02-02:
This is now implemented. Check this issue description to see what are the changes.

mvladic, 2025-02-02:
@zalexzperez In your case when you have main application screens and OTA update screens, you can set all screens (except Main) to have "Create at start" disabled. Then, in the Main screen you can have the following:

[image]

markonovic00, 2025-02-02:
Hello @mvladic  I would like to know when can be this feature used in the EEZ studio. Thanks in advance.

mvladic, 2025-02-02:
@markonovic00 This feature will be in the next release (v0.22.0) which should be out in a week or two. You can [build studio from source code](https://github.com/eez-open/studio?tab=readme-ov-file#build-and-run-from-source-all-operating-systems) if you want to try it now.

markonovic00, 2025-02-02:
I built the EEZ Studio.

I am running into an issue where the ESP32 restarts when lv_obj_del() is called. I am using simple LVGL projects and I checked the checkbox in the settings.

[image]

I had this same issue when I tried working on it alone. 

I would like to say that I am using variables on the screens so that is causing the issue maybe as well.

mvladic, 2025-02-02:
Check if some object is used after `lv_obj_del` is called - note that if screen is deleted then all the objects that are children of the that screen will also be deleted.

If you can send me your eez-project file and all the relevant source code maybe I can help you more.

markonovic00, 2025-02-02:
I disabled the tick function of the screens so there should be no usage of the data.  My steps are the following:

1. Load middle screen - Simple blank screen used for transition
2. Delete old screen
3. Create new screen
4. Load new screen

mvladic, 2025-02-03:
@markonovic00 You can't delete old screen until screen transition from old screen to new screen is done. To know when it is done you need to handle `LV_EVENT_SCREEN_UNLOADED` event for the old screen.

According to the screenshot you send I can see that your are not using EEZ Flow.  In that case, you can do something like this:

[lvgl_change_screen_no_flow.zip](https://github.com/user-attachments/files/18643493/lvgl_change_screen_no_flow.zip)

In this project only the Main screen is created at the start, and only the active screen is kept in memory. Check the `changeToScreen` in `actions.c` to see how this is implemented:

```
#include <stdint.h>

#include "actions.h"
#include "screens.h"
#include "ui.h"

////////////////////////////////////////////////////////////////////////////////

static lv_obj_t *getScreenObj(enum ScreensEnum screenId) {
    return ((lv_obj_t **)&objects)[screenId - 1];
}

static void on_screen_unloaded(lv_event_t *e) {
    if (lv_event_get_code(e) == LV_EVENT_SCREEN_UNLOADED) {
        enum ScreensEnum screenId =
            (enum ScreensEnum)(lv_uintptr_t)lv_event_get_user_data(e);
        delete_screen_by_id(screenId);
    }
}

static void deleteScreenOnUnload(enum ScreensEnum screenId) {
    lv_obj_add_event_cb(
        getScreenObj(screenId),
        on_screen_unloaded,
        LV_EVENT_SCREEN_UNLOADED,
        (void*)(lv_uintptr_t)(screenId)
    );    
}

static void changeToScreen(enum ScreensEnum screenId) {
    if (!getScreenObj(screenId)) {
        create_screen_by_id(screenId);
        if (!getScreenObj(screenId)) {
            return;
        }
        deleteScreenOnUnload(screenId);
    }

    loadScreen(screenId);
}

////////////////////////////////////////////////////////////////////////////////

void action_change_to_main_screen(lv_event_t *e) {
    changeToScreen(SCREEN_ID_MAIN);
}

void action_change_to_screen_1(lv_event_t *e) {
    changeToScreen(SCREEN_ID_SCREEN_1);
}

void action_change_to_screen_2(lv_event_t *e) {
    changeToScreen(SCREEN_ID_SCREEN_2);
}
```

mvladic, 2025-02-03:
I just made a change where I simplified  a bit the way "Delete on unload" works internally.

markonovic00, 2025-02-03:
@mvladic  This is working great. I noticed one funny thing. 

I have 3 screens, one without TabView and 2 with TabView. So here are some funny scenarios:

1. Without TabView going to screen with TabView no problem
2. TabView going to screen with TabView problem 
3. TabView to screen without TabView no problem 

It is working now, I adjusted the navigation a little bit and it is working like a charm.