eez-open/studio #710: error: ‘LV_EVENT_ROTARY’ was not declared in this scope

zhengyuanyan, 2025-01-26:
**Describe the bug**
A clear and concise description of what the bug is.

[image]

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Version [e.g. 10]

** LVGL version (if used)**

- 8.x or 9.x

**Additional context**
Add any other context about the problem here.

mvladic, 2025-01-26:
`LV_EVENT_ROTARY` doesn't exist in version 8.x, so it is not used, thus we have:

[image]

But, your version of LVGL is >= 9 and it still can't find `LV_EVENT_ROTARY`. So, my question what exact version of LVGL you are using (it is best to give me commit ID for your version).

mvladic, 2025-01-26:
I think you are using LVGL v9.0 which doesn't have `LV_EVENT_ROTARY`. Can you upgrade to v9.1 at least (latest release is V9.2.2, and current dev version is v9.3.0)?

zhengyuanyan, 2025-01-27:
git clone https://github.com/lvgl/lvgl.git

[image]

[image]

[image]

mvladic, 2025-01-27:
They changed `lv_dropdown_set_selected` API in recent dev version by adding third argument. I already fixed that in the master version of Studio.

closed 2025-02-04 by mvladic