eez-open/studio #715: [Feature] switch between open tabs with Control+Tab

akrpic77, 2025-02-05:
**Is your feature request related to a problem? Please describe.**
I have two instruments, and some shortcuts on each of them. I can press Fx keys to activate shortcut, but to switch between two instruments I have to use mouse.

**Describe the solution you'd like**
Control-Tab should switch between open tabs, switching to one used before this one would be perfect.

**Describe alternatives you've considered**
I have checked and tried many keyboard shortcuts to switch it between open tabs to no avail.

**Additional context**
using Control+Tab is generally used as a switch between context inside applications. e.g. tabs in browser.

p.s. Thank you for making this nice and useful piece of software!

mvladic, 2025-02-08:
Implemented.

- Control+Tab: Switch to next tab
- Control+Shift+Tab: Switch to previous tab

closed 2025-02-10 by mvladic

akrpic77, 2025-02-11:
I confirm that it works. Thank you.