eez-open/studio #718: [Flow] More specific error message if a flow action fails during tick execution

fietser28, 2025-02-05:
**Is your feature request related to a problem? Please describe.**
This is not a bug, but with current generic error messages generated in code by studio (in screens.c) it is very inconvenient
to determine the widget / place where the error occurs.

This is how it looked like after recent #706 testing in my project:

[image]

This text currently exist 135 times in my code. 

It would be very handy if the text would be something like: 
"Failed to evaluate Text in Label widget obj105"

Even if the widget is not explicitly named in studio it is still much easier to determine the exact widget having the problem.

**Describe the solution you'd like**
Add the object name in the error message.

**Describe alternatives you've considered**
Manual searching/deduction. Not very successful with large projects.

**Additional context**
Add any other context or screenshots about the feature request here.

mvladic, 2025-02-06:
Not all widgets have an automatically generated name, ie. only widgets that are referenced in the generated source code get an automatically generated name. So, we need some other way to identify such a widget. One idea is to call `LV_LOG_ERROR` from the location where `throwError` function is called so it prints the proper location instead of `private.cpp:691`.

fietser28, 2025-02-07:
> One idea is to call `LV_LOG_ERROR` from the location where `throwError` function is called so it prints the proper location instead of `private.cpp:691`.

This will indeed give an explicit pointer to the place in generated code and will certainly help with pinpointing the problem.

mvladic, 2025-02-08:
Implemented. 

Now, error message looks like this:

```
[Error]	(0.101, +101)	 throwError: EEZ-FLOW error: Failed to evaluate Text in Label widget | 0.4 | /home/mvladic/lvgl/v9-with-flow-no-eez-framework/src/ui/screens.c:103 eez-flow.cpp:10658
```

Two additional info are now present:

- Component path: `0.4`
- Additional source file location: `/home/mvladic/lvgl/v9-with-flow-no-eez-framework/src/ui/screens.c:103`

With component path you can find component in Studio by using Find Project Component from Edit menu:

[image]

[image]

mvladic, 2025-02-08:
The reason why I added "Find Project Component" is because not always additional source location will be present, for example if error is in LVGL action.

fietser28, 2025-02-09:
Working for me! Thank you.

closed 2025-02-10 by mvladic