eez-open/studio #722: Grey lines appears in Pages modeling

Trakeen, 2025-02-07:
**Describe the bug**
Grey lines appears in Pages modeling tool. See screenshot.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a LVGL 9.x project 240x240, circular display, dark theme, flow support
2. Go to 'Pages'
3. Click on '+'
4. Enter name 'test'
5. Add Panel to 'test'
6. Change: Style > Main part > DEFAULT > BACKGROUND > Color = #000000, BORDER > Color = #000000, Width = 0
8. Set panel size to 240x240 x=0 y=0 -> all is ok
9. Change panel height to 300 -> a grey line will appear on the right**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: OSX
 - Version: Sequoia 15.3

** LVGL version (if used)**
- 9.x

mvladic, 2025-02-07:
This gray line is scroll bar indicator. You should select the page and turn of Scrollable flag.

mvladic, 2025-02-07:
Here is animated gif that explain this:

[image]

Trakeen, 2025-02-07:
You're right, thank you

closed 2025-02-07 by Trakeen