eez-open/studio #728: LVGL include path bug

Paulo-Pecegueiro, 2025-02-12:
**Describe the bug**
When setting the LVGL include path, seems like some of the generated code do not follow this change like the eez-flow.c/.h and bitmap.c 
**To Reproduce**

Enable flow support, add any image as background, set the LVGL include to lvgl.h or something else and generate teh code.

[image]

[image]

**Desktop (please complete the following information):**
 - OS: Zorin OS
 - Version 17

** LVGL version **

- 8.x

Paulo-Pecegueiro, 2025-02-12:
@mvladic here is my proposed workaround to the end of the function generateSourceCodeForEezFramework:

https://github.com/eez-open/studio/blob/d4aae7685cdfc008824db07fe4081220d1b8f676/packages/project-editor/lvgl/build.ts#L2435

    // post fix eez-flow.h
    try {
        eezH = eezH.replace(
            `<lvgl/lvgl.h>`,
            `<${project.settings.build.lvglInclude}>`
        );
    } catch (err) {}

https://github.com/eez-open/studio/blob/d4aae7685cdfc008824db07fe4081220d1b8f676/packages/project-editor/lvgl/build.ts#L2441

    // post fix eez-flow.cpp
    try {
        let eezC = await fs.promises.readFile(
            destinationFolderPath + "/eez-flow.cpp",
            "utf-8"
        );
        eezC = eezC.replace(
            `<lvgl/lvgl.h>`,
            `<${project.settings.build.lvglInclude}>`
        );
        await fs.promises.writeFile(
            destinationFolderPath + "/eez-flow.cpp",
            eezC,
            "utf-8"
        );
    } catch (err) {}

mvladic, 2025-03-09:
Fixed.

closed 2025-03-27 by prasimix