eez-open/studio #732: AppImage cannot find Koddi module

asessa, 2025-02-17:
**Describe the bug**
Launching EEZ Studio 0.22.0 with AppImage results in an error:

A JavaScript error occurred in the main process
Uncaught Exception:
Error: Cannot find the native Koffi module; did you bundle it correctly?
    at init (/tmp/.mount_EEZ-StTZ4gn8/resources/app.asar/node_modules/koffi/index.js:466:15)
    at Object.<anonymous> (/tmp/.mount_EEZ-StTZ4gn8/resources/app.asar/node_modules/koffi/index.js:610:12)
    at Module._compile (node:internal/modules/cjs/loader:1484:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1564:10)
    at Module.load (node:internal/modules/cjs/loader:1295:32)
    at Module._load (node:internal/modules/cjs/loader:1111:12)
    at c._load (node:electron/js2c/node_init:2:16955)
    at Module.require (node:internal/modules/cjs/loader:1318:19)
    at require (node:internal/modules/helpers:179:18)
    at Object.<anonymous> (/tmp/.mount_EEZ-StTZ4gn8/resources/app.asar/build/instrument/connection/interfaces/visa-dll.js:1:929)

**To Reproduce**
Steps to reproduce the behavior:
1. Open EEZ-Studio-0.22.0.AppImage

**Expected behavior**
Should open the program.

**Desktop (please complete the following information):**
 - OS: Linux
 - Version Arch

mvladic, 2025-02-17:
Try starting like this::

```
EEZ-Studio-0.22.0.AppImage --no-sandbox
```

mvladic, 2025-02-17:
I just pushed a fix, so it will continue if koffi module load failed. BTW koffi module is required for the VISA to work, which you may not need.

asessa, 2025-02-17:
the appimage also works if rebuild from git
- git clone
- git checkout v0.22.0
- cd studio
- npm install
- npm run build

mvladic, 2025-02-17:
BTW `EEZ-Studio-0.22.0.AppImage` has `koffi` native module (koffi.node) bundled for the following platforms:

[image]

asessa, 2025-02-17:
no idea, the problem seems on app.asar, building from source works.

closed 2025-02-17 by asessa

elprocrastinador, 2025-03-04:
Please, Can you share a v0.22.1 release with that fix?
Thanks

mvladic, 2025-03-04:
You can build Studio from source. Here are the instructions: https://github.com/eez-open/studio?tab=readme-ov-file#build-and-run-from-source-all-operating-systems

elprocrastinador, 2025-03-04:
Thanks. Now it works.