eez-open/studio #735: Flow stopped with error:  Execution queue is full

mustafayondes, 2025-02-24:
[image]

Hi,

I'm working on a project using EEZ Flow. I don't use a touchscreen, so I control my screen through buttons.

I have created multiple pages, and they work fine. However, after adding some more pages with my own widget, I encountered this error:

"Flow stopped with error: Execution queue is full."

I tried increasing the execution queue size, but it didn't solve the issue.

Has anyone experienced this before? Any help would be greatly appreciated!

Thanks in advance!

mustafayondes, 2025-02-24:
Yes , I use version 0.22.0

[image]

mvladic, 2025-02-24:
Do you use any kind of loop in you flow? It can be either Loop action or something like this:

[image]

If you have a loop make sure you don't have multiple return paths to the start of loop. This can happen if you have multiple output lines for some action that is inside the loop. Something like this will full execution queue:

[image]

This should be corrected as this:

[image]

mustafayondes, 2025-02-24:
Yes I know, I do not use this kind of thing. My program is work fine with my widgets till I add more.  
However, when I try to add more widgets, I encounter an issue. If I delete one widget, I can add another one in its place, which suggests that the problem is related to the number of widgets rather than a specific widget itself.

It seems like there might be a limit affecting the number of widgets I can add. Could this be related to execution queue size or memory limitations?

mustafayondes, 2025-02-25:
> Also, check [#706](https://github.com/eez-open/studio/issues/706) if your project is getting larger and you have a problem fitting it on your hardware.

Thank you, my problem is solved. In the beginning, all pages are created, and the queue becomes full.

However, after using it for some time, the queue will fill up again, so I need to delete pages when they are unloaded. But I'm facing an issue: when I activate "Delete on Unload," the next page does not appear.

How can I fix this?

[image]

[image]

mvladic, 2025-02-25:
It is difficult to say without looking into your project. If you use "Delete on unload" then screen widgets will only be alive when screen is active. So, be sure not to work with widgets when page is inactive.