eez-open/studio #743: EEZ-flow not working on target device (ESP32S3)

janbittman, 2025-03-08:
Hi, I'm currently developing UI through EEZ studio, i got it perfectly working using basic actions, but since i found that there exist EEZ-flow i thought that this will save me a lot of time manualy handling all screen changes etc...

To the point. I got EEZ-flow working in studio run mode (for example switching screens), but when i export files(with eez-flow files) and compile them (after fixing know lvgl include bug (ignoring libs lvgl path) for EEZ-flow files) the eez-flow functionality seams not to be working at all. All other things work perfectly. Am i missing something ? 

**To Reproduce**
Create EEZ-flow project and try to flash it to ESP32S3 (waveshare 7inch touchscreen LCD)

**Expected behavior**
When i click on button in run mode in editor, it changes screen, and it should change screen as well on physical device.

 - OS: Windows 11
 -LVGL 8.x

**Additional context**
Add any other context about the problem here.

mvladic, 2025-03-08:
Do you call `ui_tick`?
https://github.com/eez-open/studio/wiki/FAQ#how-to-integrate-with-my-existing-cc-project

janbittman, 2025-03-09:
Yes man, problem solved!  I am just dumb... :D 

Thanks a lot <3 Appritiate your support and your product, it's awsome !

Jan

closed 2025-03-09 by janbittman

JanBesta, 2025-03-09:
I would like to ask you if you can switch smoothly and immediately between screens and if you have working transitions / animations, I might need some helpm because I am stuck with this. May I driectly contact you somehow? Thanks for answering.

prasimix, 2025-03-09:
> May I driectly contact you somehow? Thanks for answering.

Discord is your friend ;)