eez-open/studio #745: [LVGL] - Add slider range

derWarst, 2025-03-10:
**Is your feature request related to a problem? Please describe.**
I would like to change the range of a [NORMAL) slider with a seperate [RANGE] slider wthout having to implement it later in the code.

**Describe the solution you'd like**
Add
`[lv_slider_set_range]`
to be supported by the eez flow

mvladic, 2025-03-11:
I do not understand completely the description of this feature request:

> I would like to change the range of a [NORMAL) slider **with a seperate [RANGE] slider** without having to implement it later in the code.

I do understand what `lv_slider_set_range` function does and I can support this function from the EEZ Flow by adding a new LVGL action called "Slider Set Range" . But, the text marked as bold is the one that confuses me. If the request description is without this part it would be perfectly clear to me, but with this part it is confusing. Can you clarify this for me?

derWarst, 2025-03-11:
Ah, Ok. Sorry if my explanation was a bit unclear. 
There is a slider_1 on my main screen which will eventually control a digital potentiometer acting as a voltage divider. On a second screen, there will be a slider_2 with two knobs which can be used to adjust the range of slider_1, so I can choose to have it not go all the way to GND or V+. 
I hope this makes more sense to you.

mvladic, 2025-03-11:
I understand now. It definitely makes sense to implement "Slider Set Range" LVGL action. But, until this is implemented consider the solution presented in this project:

[lvgl_slider_range_test.zip](https://github.com/user-attachments/files/19196588/lvgl_slider_range_test.zip)

So, in this project, Slider 2 is used to define range, i.e. variables `range_min` and `range_max`. Slider 1 works with the range 0 to 100, but slider1_value is normalized (to be inside range_min to range_max) using this formula:

`range_min + slider1_value * (range_max - range_min) / 100`

So, at the end, variable `slider1_value_normalized` is always equal to the value selected in Slider 1 but within selected range given by `range_min` and `range_max` (we use Watch action to update `slider1_value_normalized`).

derWarst, 2025-03-14:
Hey, thank you for the example. I had a look at it, and it kinda works, however, I'm having problems with it, because the range of my sliders is supposed to go from -64 to 64, with the knob part starting at 0. I'm trying to fiddle around with the math functions, but so far can't figure out how to make your example work for my case.

mvladic, 2025-03-14:
I'm not sure I understand completely your use case . But, with my example I demonstrated that any range can be supported with the slider with some math. You can try to explain me more precisely your use case and I might help.

derWarst, 2025-03-14:
Here's a picture what I want it to look like:

[image]

Both sliders should have a range of 128 steps, from -64 to 64, with zero in the middle. 
I can't figure out the math to have it work with negative and positive values.

mvladic, 2025-03-14:
Here is the formula: 

```range_min + ( slider1_value - (-64) ) * (range_max - range_min) / ( 64 - (-64) )```

or:

```range_min + (slider1_value + 64) * (range_max - range_min) / 128```

And here is the project:

[lv_slider_range_test.zip](https://github.com/user-attachments/files/19252936/lv_slider_range_test.zip)

derWarst, 2025-03-14:
I see, thank you. 
I worked a bit with the project and made the sliders work in a way so slider 2 changes the width of slider 1, which is the way I want it to behave in my Interface. 
But now I'm stuck again with trying to figure out how to make the value of slider1 stay the same while setting the range. So it is only moved by touching slider 1 or if it touches the range boundaries.

[lv_slider_range_test.zip](https://github.com/user-attachments/files/19253693/lv_slider_range_test.zip)

mvladic, 2025-03-14:
This is little bit tricky, so I'm changing the approach, i.e. I'm not using Watch but VALUE_CHANGED event:

[lv_slider_range_tesst.zip](https://github.com/user-attachments/files/19254378/lv_slider_range_tesst.zip)

I think you now how all the ingredients to continue tweaking this on your own 😉

mvladic, 2025-03-15:
Implemented.

You can now specify slider range using Expressions properties:

[image]

or using LVGL action:

[image]

derWarst, 2025-03-16:
YES! Thank you so much!

closed 2025-03-27 by prasimix