eez-open/studio #748: EEZ versus SLS

klaatu2a, 2025-03-12:
Looking for a UI solution which does not require coding expertise
What is your view of EEZ versus SLS?
I asked a couple of AI to compare them,. and its paints EEZ as a low end UI designer and SLS has a more advanced and feature reach.  It may just be that EEZ is in its infancy. 
AI states that support for both is poor.
Pricing on SLS is a barrier and might explain why there is little to no support community.
EEZ using discord as a community forum, does not seem applicable.

SinglWolf, 2025-03-15:
`SLS` support via the forum has not responded for weeks. While waiting for a response from `SLS` support, I decided to explore `EEZ`.
`SLS` does not have the functionality that `EEZ` has. Eg, in `SLS` for a `checkbox`, it is not possible to change the width of the field from the `checkbox` to the text label, in `EEZ` it is possible. Also, in `EEZ`, can set the location of the `checkbox` to the left or right of the text label.
I've only been studying` EEZ` for a couple of evenings and the impression is only positive. Let's see what kind of support is here 😺.

klaatu2a, 2025-03-15:
SL is coming out with a wholly new direction which they call SL Vision.  I signed up for the beta program, but they have not released the trial software yet.

I have not spent much time looking into EEZ yet.  But so far, I find discord just a horrid vehicle for community.  And so far you are the first response in 2 days on github.  Not a good sign

SinglWolf, 2025-03-15:
> SL is coming out with a wholly new direction which they call SL Vision. I signed up for the beta program, but they have not released the trial software yet.

I'm also waiting for SL Vision. 
> I have not spent much time looking into EEZ yet.

My project is idle due to an error in SLS. I'll try to port my project to EEZ, let's see what happens.
> And so far you are the first response in 2 days on github. Not a good sign

Oh, there are two of us already! It's not that bad! 😺

SinglWolf, 2025-03-15:
> But so far, I find discord just a horrid vehicle for community.

Can discuss it here https://github.com/eez-open/studio/discussions

SinglWolf, 2025-03-16:
> And so far you are the first response in 2 days on github. Not a good sign

Look at the speed with which support not only responded to me, but also made [improvements](https://github.com/eez-open/studio/issues/752) and [fixed bugs](https://github.com/eez-open/studio/issues/751)!

closed 2025-03-25 by mvladic