eez-open/studio #751: When creating font, symbol mapping range not working?

SinglWolf, 2025-03-15:
**Describe the bug**
I am an old (and not very satisfied) `SLS` user who is looking for a decent alternative.
I'm currently trying to port my project to `EEZ`.
**To Reproduce**
I am creating set icons from font with mapping to numeric characters for use from `LVGL`.
[image]
There were no error messages when creating the font.

**Expected behavior**
Expected for (one symbol) eg `0xf690=>0x34`
```
    /* U+0034 "4" */
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x12, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0xe, 0xf2, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0xff, 0x40, 0x0,
    0x0, 0x0, 0x1, 0x20, 0xf, 0xf4, 0x0, 0x0,
    0x0, 0x0, 0xdf, 0x30, 0xff, 0x40, 0x0, 0x0,
    0x0, 0xe, 0xf5, 0xf, 0xf4, 0x0, 0x0, 0x2,
    0x10, 0xef, 0x50, 0xff, 0x40, 0x0, 0x3, 0xfd,
    0xe, 0xf5, 0xf, 0xf4, 0x0, 0x0, 0x5f, 0xe0,
    0xef, 0x50, 0xff, 0x40, 0x21, 0x5, 0xfe, 0xe,
    0xf5, 0xf, 0xf4, 0x2f, 0xe0, 0x5f, 0xe0, 0xef,
    0x50, 0xff, 0x43, 0xff, 0x5, 0xfe, 0xe, 0xf4,
    0xf, 0xf3, 0x9, 0x70, 0x9, 0x60, 0x69, 0x0,
    0x79, 0x0,
```
In fact, it turns out like this
```
    /* U+F690 "" */
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x12, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0xe, 0xf2, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0xff, 0x40, 0x0,
    0x0, 0x0, 0x1, 0x20, 0xf, 0xf4, 0x0, 0x0,
    0x0, 0x0, 0xdf, 0x30, 0xff, 0x40, 0x0, 0x0,
    0x0, 0xe, 0xf5, 0xf, 0xf4, 0x0, 0x0, 0x2,
    0x10, 0xef, 0x50, 0xff, 0x40, 0x0, 0x3, 0xfd,
    0xe, 0xf5, 0xf, 0xf4, 0x0, 0x0, 0x5f, 0xe0,
    0xef, 0x50, 0xff, 0x40, 0x21, 0x5, 0xfe, 0xe,
    0xf5, 0xf, 0xf4, 0x2f, 0xe0, 0x5f, 0xe0, 0xef,
    0x50, 0xff, 0x43, 0xff, 0x5, 0xfe, 0xe, 0xf4,
    0xf, 0xf3, 0x9, 0x70, 0x9, 0x60, 0x69, 0x0,
    0x79, 0x0,
```

**Desktop (please complete the following information):**
 - OS: 
 Windows
 - Version
10

** LVGL version (if used)**
9.2.2

Why is that?

mvladic, 2025-03-16:
This feature is now implemented.

SinglWolf, 2025-03-16:
[image]
It's working! Thanks! 👍

closed 2025-03-27 by prasimix