eez-open/studio #752: [LVGL] Choosing how to use fonts

SinglWolf, 2025-03-16:
I've studied (https://github.com/eez-open/studio/issues/674), but I still haven't found a way to create a font in binary format. Only a file is created, eg `ui_font_main_12.c`.
I'm trying to migrate my project from `SLS`. In `SLS` two files are generated immediately when creating a font. The first is in the form of source code, and the second is in binary format `LVGL`.
I copy the binary font file to the sd card and initialize the font like eg this:
```c
lv_font_t *ui_font_icons = lv_binfont_create("L:/fonts/Sum12.bin");
```
For example, for a text label, this font is used like this:
```c
lv_obj_t *ui_main = lv_obj_create(NULL);
lv_obj_t *ui_PlayState = lv_label_create(ui_main);
lv_obj_set_style_text_font(ui_PlayState, ui_font_icons, LV_PART_MAIN | LV_STATE_DEFAULT);
```
It would also be great if, when using a binary font file, a replacement font was added, provided that a fallback font was assigned when creating the main font, for example, like this:
```c
lv_font_t *ui_font_icons = lv_binfont_create("L:/fonts/Sum12.bin");
ui_font_icons->fallback = &lv_font_montserrat_12;
```

The condition for choosing to use the binary format can be used in this way.
If the user selects a file system driver from this menu and assigns a drive letter, then the binary format is considered to be used. (The image was get from the `SLS` project settings.)

[image]
This menu is linked to the `LVGL` settings file `lv_conf.h`
```c
/*File system interfaces for common APIs */

/*API for fopen, fread, etc*/
#define LV_USE_FS_STDIO 0
#if LV_USE_FS_STDIO
    #define LV_FS_STDIO_LETTER 'A' //'\0'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_STDIO_PATH "drive/"         /*Set the working directory. File/directory paths will be appended to it.*/
    #define LV_FS_STDIO_CACHE_SIZE 0    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif

/*API for open, read, etc*/
#define LV_USE_FS_POSIX 1
#if LV_USE_FS_POSIX
    #define LV_FS_POSIX_LETTER 'L'
    #define LV_FS_POSIX_PATH "sd-card"
    #define LV_FS_POSIX_CACHE_SIZE 0    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif

/*API for CreateFile, ReadFile, etc*/
#define LV_USE_FS_WIN32 0
#if LV_USE_FS_WIN32
    #define LV_FS_WIN32_LETTER '\0'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_WIN32_PATH ""         /*Set the working directory. File/directory paths will be appended to it.*/
    #define LV_FS_WIN32_CACHE_SIZE 0    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif

/*API for FATFS (needs to be added separately). Uses f_open, f_read, etc*/
#define LV_USE_FS_FATFS 0
#if LV_USE_FS_FATFS
    #define LV_FS_FATFS_LETTER '\0'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_FATFS_CACHE_SIZE 0    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif
```

mvladic, 2025-03-16:
We can add the following settings for the fonts in the project setting - Build:

```
Store fonts in filesystem        [x]
Fonts filesystem path            L:/fonts/_____________
```

When "Store fonts in filesystem" is enabled then .bin files will be created and `lv_font_t *<font_name> = lv_binfont_create("L:/fonts/<font_name>.bin");` will be added to the `ui_init`.

mvladic, 2025-03-16:
This is now implemented. Can you please [build Studio from source](https://github.com/eez-open/studio?tab=readme-ov-file#build-and-run-from-source-all-operating-systems) and test it?

Feature to specify fallback font already exists (see https://github.com/eez-open/studio/issues/620#issuecomment-2466280070).

Here is how I tested:

[image]

Change in `lv_conf.h`:

```C
/*API for open, read, etc*/
#define LV_USE_FS_POSIX 1
#if LV_USE_FS_POSIX
    #define LV_FS_POSIX_LETTER 'L'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_POSIX_PATH "/home/mvladic/lvgl/v9-with-flow-no-eez-framework"         /*Set the working directory. File/directory paths will be appended to it.*/
    #define LV_FS_POSIX_CACHE_SIZE 0    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif
```

So, the location of the font file is: `/home/mvladic/lvgl/v9-with-flow-no-eez-framework/src/ui/ui_font_oswald_16.bin`.

The generated code looks like this:

```
    lv_font_t *ui_font_oswald_16 = lv_binfont_create("L:/src/ui/ui_font_oswald_16.bin");
    ui_font_oswald_16->fallback = &lv_font_montserrat_24;
```

[image]

mvladic, 2025-03-16:
We will need a similar feature for the images also. I'm not sure right now how this will work, so any help is appreciated.

SinglWolf, 2025-03-16:
> This is now implemented. Can you please [build Studio from source](https://github.com/eez-open/studio?tab=readme-ov-file#build-and-run-from-source-all-operating-systems) and test it?

This is incredibly great news!
I have Windows 10. Are there any difficulties I may encounter when building from source code?

> We will need a similar feature for the images also. I'm not sure right now how this will work, so any help is appreciated.

I will provide any possible assistance! As soon as EEZ Studio is build I will let you know how the testing went.

mvladic, 2025-03-16:
> I have Windows 10. Are there any difficulties I may encounter when building from source code?

I'm also on Windows. Currently, I'm using NodeJS version 20.17.0 without problems.

SinglWolf, 2025-03-16:
At first glance, the code is generated as stated, but something is missing.
Can make the font path and image path as a constant?
```c
const char * ui_img_boot_png = "L:/assets/boot.png";
const char * ui_icon12_path = "L:/assets/ui_font_icon12.bin";
// or macro
#define ICON_12_PATH "L:/assets/ui_font_icon12.bin"

```
For now, I will continue porting my project to EEZ Studio.
I'll let you know if my have any new ideas. Thank you for your quick response to my inquiries!!! I hope that my project will work soon.

mvladic, 2025-03-16:
> Can make the font path and image path as a constant?

Why you need this?

SinglWolf, 2025-03-16:
> Why you need this?

Eg
```c
#if defined(ILI9341_DRIVER)
const char  *ui_icon_path = "L:/assets/ui_font_icon12.bin";
#elif defined(ILI9488_DRIVER)
const char *ui_icon_path = "L:/assets/ui_font_icon16.bin";
#endif
```
Also need to make it possible to select the type of file system. As here:
[image]
For example, can write own callbacks for `POSIX`, as I have done.
For `STDIO` is used built-in driver.

SinglWolf, 2025-03-16:
> I'm also on Windows. Currently, I'm using NodeJS version 20.17.0 without problems with building.

Hmm, I managed to build the file `EEZ Studio Setup 0.23.0.exe` using command `npm run dist` only at startup `cmd.exe` on behalf of the administrator. Need to specify this in the instructions, I think.

mvladic, 2025-03-16:
> Hmm, I managed to build the file EEZ Studio Setup 0.23.0.exe using command npm run dist only at startup cmd.exe on behalf of the administrator. Need to specify this in the instructions, I think.

What error do you get when your run `npm run dist` without running `cmd.exe` as administrator?

mvladic, 2025-03-16:
> Also need to make it possible to select the type of file system. As here:
> For example, can write own callbacks for POSIX, as I have done.
> For STDIO is used built-in driver.

I don't understand why Studio needs to know details how filesystem driver is initialized.

SinglWolf, 2025-03-16:
> I don't understand why Studio needs to know details how filesystem driver is initialized.

This is necessary for `LVGL`. You've made changes for `POSIX` in `lv_conf.h`. 
```c
/*API for open, read, etc*/
#define LV_USE_FS_POSIX 1
#if LV_USE_FS_POSIX
    #define LV_FS_POSIX_LETTER 'L'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_POSIX_PATH "/home/mvladic/lvgl/v9-with-flow-no-eez-framework"         /*Set the working directory. File/directory paths will be appended to it.*/
    #define LV_FS_POSIX_CACHE_SIZE 0    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif
```
Why not make this feature available for other types of drivers? The `SLS` developers created such a menu for a reason.

> What error do you get when your run `npm run dist` without running `cmd.exe` as administrator?

When I ran the `npm run dist` command again on behalf of the user, everything went smoothly. But I remember that at the first launch, the error was related to the inability to download or unzip the file `electron-v32.2.2-win32-x64.zip` on the path `C:\Users\Alexander\AppData\Local\electron\Cache\`. After running `npm run dist` as an administrator, this problem was resolved and now `npm run dist` can be run as a user. There were several other similar errors for other files, the names of which, unfortunately, I did not remember.

SinglWolf, 2025-03-17:
I don't have a Windows computer at home. Only Linux ROSA FRESH 13 Plasma 6. I decided to build EEZ Studio from the source code, but did not progress beyond the `npm install` command.
The last few lines of console
```bash
gyp verb node dev dir /home/alex/.electron-gyp/32.2.2
gyp verb python /usr/bin/python3
Error: not found: make
    at getNotFoundError (/home/alex/123/node_modules/which/which.js:10:17)
    at /home/alex/123/node_modules/which/which.js:57:18
    at new Promise (<anonymous>)
    at step (/home/alex/123/node_modules/which/which.js:54:21)
    at /home/alex/123/node_modules/which/which.js:71:22
    at new Promise (<anonymous>)
    at subStep (/home/alex/123/node_modules/which/which.js:69:33)
    at /home/alex/123/node_modules/which/which.js:80:22
    at /home/alex/123/node_modules/isexe/index.js:42:5
    at /home/alex/123/node_modules/isexe/mode.js:8:5
    at FSReqCallback.oncomplete (node:fs:199:21) {
  code: 'ENOENT'
}

✖ Rebuild Failed

An unhandled error occurred inside electron-rebuild
node-gyp failed to rebuild '/home/alex/123/node_modules/@serialport/bindings-cpp'

Error: node-gyp failed to rebuild '/home/alex/123/node_modules/@serialport/bindings-cpp'
    at ChildProcess.<anonymous> (/home/alex/123/node_modules/@electron/rebuild/lib/module-type/node-gyp/node-gyp.js:121:24)
    at ChildProcess.emit (node:events:507:28)
    at ChildProcess._handle.onexit (node:internal/child_process:294:12)
npm error code 255
npm error path /home/alex/123
npm error command failed
npm error command sh -c electron-rebuild
npm error A complete log of this run can be found in: /home/alex/.npm/_logs/2025-03-17T13_22_31_351Z-debug-0.log
```
The last few lines of `2025-03-17T13_22_31_351Z-debug-0.log`
```bash
1577 silly packumentCache corgi:https://registry.npmjs.org/gulp cache-miss
1578 http cache https://registry.npmjs.org/gulp 32ms (cache hit)
1579 silly packumentCache corgi:https://registry.npmjs.org/gulp set size:60484 disposed:false
1580 info run @serialport/bindings-cpp@12.0.1 install node_modules/@serialport/bindings-cpp node-gyp-build
1581 info run better-sqlite3@11.5.0 install node_modules/better-sqlite3 prebuild-install || node-gyp rebuild --release
1582 info run koffi@2.9.1 install node_modules/koffi node src/cnoke/cnoke.js -p . -d src/koffi --prebuild
1583 info run @serialport/bindings-cpp@12.0.1 install { code: 0, signal: null }
1584 info run usb@2.14.0 install node_modules/usb node-gyp-build
1585 info run koffi@2.9.1 install { code: 0, signal: null }
1586 info run usb@2.14.0 install { code: 0, signal: null }
1587 info run better-sqlite3@11.5.0 install { code: 0, signal: null }
1588 info run electron@32.2.2 postinstall node_modules/electron node install.js
1589 info run es5-ext@0.10.64 postinstall node_modules/es5-ext  node -e "try{require('./_postinstall')}catch(e){}" || exit 0
1590 info run es5-ext@0.10.64 postinstall { code: 0, signal: null }
1591 info run electron@32.2.2 postinstall { code: 0, signal: null }
1592 verbose stack Error: command failed
1592 verbose stack     at promiseSpawn (/home/alex/.nvm/versions/node/v23.10.0/lib/node_modules/npm/node_modules/@npmcli/promise-spawn/lib/index.js:22:22)
1592 verbose stack     at spawnWithShell (/home/alex/.nvm/versions/node/v23.10.0/lib/node_modules/npm/node_modules/@npmcli/promise-spawn/lib/index.js:124:10)
1592 verbose stack     at promiseSpawn (/home/alex/.nvm/versions/node/v23.10.0/lib/node_modules/npm/node_modules/@npmcli/promise-spawn/lib/index.js:12:12)
1592 verbose stack     at runScriptPkg (/home/alex/.nvm/versions/node/v23.10.0/lib/node_modules/npm/node_modules/@npmcli/run-script/lib/run-script-pkg.js:77:13)
1592 verbose stack     at runScript (/home/alex/.nvm/versions/node/v23.10.0/lib/node_modules/npm/node_modules/@npmcli/run-script/lib/run-script.js:12:10)
1592 verbose stack     at async Install.exec (/home/alex/.nvm/versions/node/v23.10.0/lib/node_modules/npm/lib/commands/install.js:163:9)
1592 verbose stack     at async Npm.exec (/home/alex/.nvm/versions/node/v23.10.0/lib/node_modules/npm/lib/npm.js:207:9)
1592 verbose stack     at async module.exports (/home/alex/.nvm/versions/node/v23.10.0/lib/node_modules/npm/lib/cli/entry.js:74:5)
1593 verbose pkgid eezstudio@0.23.0
1594 error code 255
1595 error path /home/alex/123
1596 error command failed
1597 error command sh -c electron-rebuild
1598 verbose cwd /home/alex/123
1599 verbose os Linux 6.12.13-generic-4rosa13-x86_64
1600 verbose node v23.10.0
1601 verbose npm  v10.9.2
1602 verbose exit 255
1603 verbose code 255
1604 error A complete log of this run can be found in: /home/alex/.npm/_logs/2025-03-17T13_22_31_351Z-debug-0.log
```
In my opinion, is it the same problem as in Windows with `electron-rebuild`?

SinglWolf, 2025-03-18:
After a long search and a sleepless night, my managed to find all the dependencies for `Linux ROSA FRESH 13 x86_64`.
```bash
sudo dnf install make gcc-c++ rpm-build lib64udev-devel lib64nss3
```
Please add this to instruction.

SinglWolf, 2025-03-19:
I suggest using instead of
[image]
Make Switch `Use File System` For both fonts and images.
If selected, show:

1. A drop-down list with uppercase drive letters from `A` to `Z`. Eg, `C:/`
2. The text field "Font and image common storage folder name". Eg `sd-card`. Creating a folder with this name in the root directory of the project.
 [image]
3. The text field `Folder name for fonts`. Eg, "fonts".
4. The text field `Folder name for images`. Eg, "img".
5. Creating these folders in the directory `sd-card`
[image]
6. Exporting fonts to these directories in binary form (already done), and images in their original form, eg. "*.png", "*.jpg".
7. Create paths to images "C:/img" and fonts "C:/fonts" as string constants, for example 
``` c
const char *ui_img_boot_png = "C:/img/boot.png";
const char *main12 = "C:/fonts/main12.bin";

```
8. Creating choice to assign a string variable to the font and image, as here
[image]
### In order for LVGL to understand image paths correctly, need to make changes to lv_conf.h, eg
```c
/*LODEPNG decoder library*/
#define LV_USE_LODEPNG 1

/*PNG decoder(libpng) library*/
#define LV_USE_LIBPNG 0

/*BMP decoder library*/
#define LV_USE_BMP 0

/* JPG + split JPG decoder library.
 * Split JPG is a custom format optimized for embedded systems. */
#define LV_USE_TJPGD 1

/* libjpeg-turbo decoder library.
 * Supports complete JPEG specifications and high-performance JPEG decoding. */
#define LV_USE_LIBJPEG_TURBO 0

/*GIF decoder library*/
#define LV_USE_GIF 1 //0
#if LV_USE_GIF
/*GIF decoder accelerate*/
#define LV_GIF_CACHE_DECODE_DATA (16 * 1024)
#endif

```
I hope it will be clear. I tried 😺

mvladic, 2025-03-23:
Now, both Images and Fonts can be exported in binary format for use with external storage (file system).

[image]

When "Image export mode" or "Font export mode" is set to "Binary" then "File system path" is used to form file path for the image and font.

Here is the example how image is referenced in generated code:

```
lv_image_set_src(obj, "L:/src/ui/ui_image_settings.bin");
```

and here is the example how font is created:

```
ui_font_oswald_24 = lv_binfont_create("L:/src/ui/ui_font_oswald_24.bin");
```

When file system is initialized in the `lv_conf.h` like this:

```C
#define LV_USE_FS_POSIX 1
#if LV_USE_FS_POSIX
    #define LV_FS_POSIX_LETTER 'L'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_POSIX_PATH "/home/mvladic/lvgl/lv_port_pc_vscode/main"         /*Set the working directory. File/directory paths will be appended to it.*/
    #define LV_FS_POSIX_CACHE_SIZE 0    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif
```

then final file paths are:

```
/home/mvladic/lvgl/lv_port_pc_vscode/main/src/ui/ui_image_settings.bin
/home/mvladic/lvgl/lv_port_pc_vscode/main/src/ui/ui_font_oswald_24.bin
```

I think this is a good start. We can later add some of the stuff that @SinglWolf recommended if required for the real word use case.

SinglWolf, 2025-03-23:
> We can later add some of the stuff that [@SinglWolf](https://github.com/SinglWolf) recommended if required for the real word use case.

What I recommended to implement in EEZ Studio is a real use in my project and it works!!!!

SinglWolf, 2025-03-24:
I rebuilt EEZ Studio, and here is the result in a project without a "Support Flow"
Source code is clearly from the `eez-framework`
[image]
[image]
I did not find any links to `extern const ext_img_desc_t images[4]`
[image]

SinglWolf, 2025-03-24:
> When file system is initialized in the `lv_conf.h` like this:
> 
> #define LV_USE_FS_POSIX 1
> #if LV_USE_FS_POSIX
>     #define LV_FS_POSIX_LETTER 'L'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
>     #define LV_FS_POSIX_PATH "/home/mvladic/lvgl/lv_port_pc_vscode/main"         /*Set the working directory. File/directory paths will be appended to it.*/
>     #define LV_FS_POSIX_CACHE_SIZE 0    /*>0 to cache this number of bytes in lv_fs_read()*/
> #endif

On a real device, for example, on an SD card, there  path for `LV_FS_POSIX_PATH` can't be that long!
Defining the path for `LV_FS_POSIX_PATH` is necessary if, for example, the beginning of the path was skipped in `lv_image_set_src(obj, "L:ui_image_boot_ru.bin")`. Then `LVGL` will it will automatically get `/sd-card/` from at `LV_FS_POSIX_PATH`, and then you will get "L:/sd-card/ui_image_boot_ru.bin".

mvladic, 2025-03-24:
> I rebuilt EEZ Studio, and here is the result in a project without a Support Flow
> Source code is clearly from the eez-framework.

No, it is not from the eez-framework. Did you actually try to compile generated source code?

mvladic, 2025-03-24:
> On a real device, for example, on an SD card, there path for LV_FS_POSIX_PATH can't be that long!
Defining the path for LV_FS_POSIX_PATH is necessary if, for example, the beginning of the path was skipped in lv_image_set_src(obj, "L:ui_image_boot_ru.bin"). Then LVGL will it will automatically get /sd-card/ from at #define LV_FS_POSIX_PATH "/sd-card/", and then you will get in out "L:/sd-card/ui_image_boot_ru.bin".
https://docs.lvgl.io/9.2/libs/fs.html#usage

I did compiled and tested all of this with success using following repository as starting point: https://github.com/lvgl/lv_port_pc_vscode. I didn't try on any "real" device, but I don't see any reason why it shouldn't work.

closed 2025-03-27 by prasimix

SinglWolf, 2025-03-27:
Why did they close this issue? There's still a lot to decide!

prasimix, 2025-03-28:
> [@mvladic](https://github.com/mvladic), why close this issue? There's still a lot to decide!

It is closed because it includes changes added in the v0.23 release. Feel free to open a new issue with a reference to this one where you can list whatever is on your mind.