eez-open/studio #754: Problems with theme colors

SinglWolf, 2025-03-17:
The colors are not duplicated or copied, the color is always black.
[image]
Пипетка неправильно определяет цвет
[image]
It should be pure green.
[image]
There is no context menu for right-click (copy, paste, cut)
[image]

mvladic, 2025-03-23:
> The colors are not duplicated or copied, the color is always black.

This should be fixed now.

> The pipette does not detect the color correctly
> It should be pure green.

I can't reproduce this.

> There is no context menu for right-click (copy, paste, cut)

Unfortunately, the built in Chrome color picker (https://developer.mozilla.org/en-US/docs/Web/HTML/Element/input/color) doesn't show context menu. Only solution is to use alternative color picker instead of built in,

SinglWolf, 2025-03-24:
> I can't reproduce this.

[Windows]
I choose a "color" with a pure green color

[image]

Then I point the pipette at the current "color". It is immediately obvious that the color is not pure green.

[image]

Result

[image]

An pipette from another location

[image]

[image]
Is something wrong with Windows?

mvladic, 2025-03-24:
> There is no shadow here

The green color of the digits in your screenshot is exactly: `RGB(120, 252, 56)`, which I verified using Paint.NET program:

[image]

So, I'm still not sure what is exactly the problem here.

> Does it turn out that there is no practical benefit from the eyedropper in EEZ Studio? In SLS, everything works without "applying shadows".

As I already said, this is the standard Color picker of the Chrome browser (as you may or may not know, Studio is written using [Electron](https://www.electronjs.org/), which is based on Chromium) and eyedropper is one of its features. You don't need to use it to pick the color.

SinglWolf, 2025-03-24:
> The green color of the digits in your screenshot is exactly: `RGB(120, 252, 56)`, which I verified using Paint.NET program:

Please excuse me. This screenshot was from EEZ Studio's previous commits on Windows. I have now checked the current build on Linux - the eyedropper is working. I'll build EEZ Studio on Windows tomorrow and check eyedropper out.
I honestly admit my mistake and apologize.

SinglWolf, 2025-03-25:
I'm just in shock. The pipette does not work on Linux again. I started thinking about why was working yesterday. Before the pipette started working, I had the SLS application running. I decided to repeat it. I started SLS and switched to Studio. Get this, empty screen
[image]
I did a reload via `CTRL+R`, the screen appeared and the pipette started working again! After closing `SLS`, the screen in `EEZ Studio` is blank again and the pipette stopped working.
Everything is the same in `Windows`. The pipette does not work. Running `SLS` does not affect anything.
I'm sad.
But I wonder how `SLS` influenced the eyedropper in `EEZ Studio` for `Linux`?

SinglWolf, 2025-03-25:
I decided to set the border color of the container
[image]
When trying to drag the color from Themes tab to propertis a border color field, it turns out this
[image]

mvladic, 2025-03-25:
> When trying to drag the color from Themes tab to propertis a border color field, it turns out this

Fixed.