eez-open/studio #755: [LVGL] No way to change the style of the TabView widget.

SinglWolf, 2025-03-18:
I did not find how to change the color of the text and other parameters on the tabs of the `Tabview` widget. There is such an opportunity in `SLS`.

[image]

[image]

mvladic, 2025-03-18:
One of the official examples is Styled Tabview:

[image]

where you can see how Tabview can be styled. The idea is to first add two Containers to the Tab, one for Bar and other for Content, and Tabs should be later added to the Content:

[image]

SinglWolf, 2025-03-18:
It never even occurred to me to look at examples. I sincerely apologize.
Judging by the picture, this is a rather original method. I will try. Closing the issue.

closed 2025-03-18 by SinglWolf

fermion-boson, 2025-06-20:
But I still can't change the blue of the tab