eez-open/studio #756: Image placed inside of any container is no longer visible

davidrfloydii, 2025-03-18:
**Describe the bug**
Any time an image or an image button or a button with an image is placed inside of a container it is no longer visible. This has been tested with panels, containers, and widgets. All produce the same result which is the image is no longer visible when executing on Arduino. Displays fine in the GUI but not during runtime. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Screen'
2. Click on 'Add image component'
3. Run project and observe the image appear
4. Go back to 'Screen'
5. Click on 'Add panel'
6. Drag image into 'Panel'
7. Run project and observe the image disappear

**Expected behavior**
The image is expected to be visible in the runtime as it is within the GUI.

**Desktop (please complete the following information):**
 - OS: Mac
 - Version 15.3.2

** LVGL version (if used)**
- 8.x

**Additional context**
I've tested this extensively inside of widgets, outside of widgets and it is very repeatable. Unless the image is a root level element it is not visible during runtime. If it is inside a button or is an image button the functionality of that event will work. It simply doesn't display correctly.

mvladic, 2025-03-18:
Make sure Image property Top (y) and Left (x) are within Panel's visible area.

[image]

davidrfloydii, 2025-03-18:
> Make sure Image property Top (y) and Left (x) are within Panel's visible area.
> 
> [image]

Thank you I had that set and it didn't work in the GUI it was showing in the correct location and visible. However, I did figure out how to solve it. I had to explicitly set an image style to recolor and set opacity. This resolves the issue, although it still is not clear why I have to recolor an image to the same color it is. But it works that way. Thank you!

closed 2025-03-25 by mvladic

davidrfloydii, 2025-03-27:
Just FYI the left/top was not my issue. I did figure it out though apparently when using 1 bit color images there can be recoloring by default. This was causing the image to not be visible since it recolored to black and was going onto a black background. The image itself was white. Just FYI in case others experience this.