eez-open/studio #758: [LVGL]  Can LVGL play GIFs in EEZ?

yirujikeshenshihao, 2025-03-19:
Can LVGL play GIFs in EEZ?
I found that it can be imported, but it cannot be played.
May I ask who knows the correct way to set it up?
If it doesn't work in the current version, can we add GIFs in the next update?

mvladic, 2025-03-23:
I think LVGL has no support for the animated GIF. You can use AnimationImage widget instead.

yirujikeshenshihao, 2025-03-24:
> I think LVGL has no support for the animated GIF. You can use AnimationImage widget instead.
When I have a lot of pictures, I can only set them to AnimationImage one by one, which is a waste of time. Is there an option to increase batch size?

SinglWolf, 2025-03-24:
> I think LVGL has no support for the animated GIF. You can use AnimationImage widget instead.

You're completely wrong. Check out [GIF decoder](https://docs.lvgl.io/9.2/libs/gif.html#), there's even an example with gif animation.

yirujikeshenshihao, 2025-03-24:
> > I think LVGL has no support for the animated GIF. You can use AnimationImage widget instead.
> 
> You're completely wrong. Check out [GIF decoder](https://docs.lvgl.io/9.2/libs/gif.html#), there's even an example with gif animation.
Yes, I noticed this example, so I suggested that FFmpeg could also be used. If EEZ can support it, that would be really great

mvladic, 2025-03-24:
> When I have a lot of pictures, I can only set them to AnimationImage one by one, which is a waste of time. Is there an option to increase batch size?

Sorry, there is no such option currently.

yirujikeshenshihao, 2025-03-25:
> > When I have a lot of pictures, I can only set them to AnimationImage one by one, which is a waste of time. Is there an option to increase batch size?
> 
> Sorry, there is no such option currently.

I understand, can we add this feature?

Vadimatorik, 2025-03-25:
> > I think LVGL has no support for the animated GIF. You can use AnimationImage widget instead.
> 
> You're completely wrong. Check out [GIF decoder](https://docs.lvgl.io/9.2/libs/gif.html#), there's even an example with gif animation.

We encountered the same problem on our project. The difference between adding png and gif is only in two lines of code. At the moment, we are waiting for a solution to this problem. If it is not there, then we will have to write a script to change 2 lines for each file in the generated project (I really hope this functionality will be added)

mvladic, 2025-03-25:
> We encountered the same problem on our project. The difference between adding png and gif is only in two lines of code. At the moment, we are waiting for a solution to this problem. If it is not [...]

To better understand the required solution, can you give me more details what 2 lines you need to change?

mvladic, 2025-03-25:
> I understand, can we add this feature?

Yes, but can you please create another issue (feature request) for adding multiple images to the AnimationImage widget as a single operation.

yirujikeshenshihao, 2025-03-25:
> > 我明白了，我们可以添加这个功能吗？
> 
> Yes, but can you please create another issue (feature request) for adding multiple images to the AnimationImage widget as a single operation.

Of course, thank you for your help

yirujikeshenshihao, 2025-03-25:
> > I understand, can we add this feature?
> 
> Yes, but can you please create another issue (feature request) for adding multiple images to the AnimationImage widget as a single operation.

Regarding my other question, could you please add a [QR code](https://docs.lvgl.io/8.3/libs/qrcode.html)?

petrovichkr, 2025-03-26:
> To better understand the required solution, can you give me more details what 2 lines you need to change?

Workaround step-by-step:

1. In EEZ Studio project: add gif image (say `sample_gif.gif`) and set its color format to `RAW`
2. Add Image widget into screen and set image source to `sample_gif`
3. Build EEZ Studio project. In generated `screens.c` we get following code:
```c
void create_some_screen() {
    ...
    lv_obj_t *obj = lv_obj_create(0);
    ...
    {
        lv_obj_t *parent_obj = obj;
        ...
        {
            lv_obj_t *obj = lv_img_create(parent_obj);
            lv_obj_set_pos(obj, 5, 230);
            lv_obj_set_size(obj, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
            lv_img_set_src(obj, &img_sample_gif);
        }
        ...
    }
    ...
}
```
4. Replace `lv_img_create()` by `lv_gif_create()` and `lv_img_set_src()` by `lv_gif_set_src()`:
```c
void create_some_screen() {
    ...
    lv_obj_t *obj = lv_obj_create(0);
    ...
    {
        lv_obj_t *parent_obj = obj;
        ...
        {
            lv_obj_t *obj = lv_gif_create(parent_obj); // img->gif
            lv_obj_set_pos(obj, 5, 230);
            lv_obj_set_size(obj, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
            lv_gif_set_src(obj, &img_sample_gif); // img->gif
        }
        ...
    }
    ...
}
```
5. Don't forget about `LV_USE_GIF` set to 1 in `lv_conf.h`

ASEdkt, 2025-12-12:
@petrovichkr 
I have applied your fix however I'm facing issue regarding palette not found, and it seems there is no color palette in the generated GIF in RAW format

[image]

[image]

[image]

IWILLTBEST, 2026-09-09:
Hi @mvladic — following up on your earlier "We will add Gif widget in the future": we would be happy to contribute it, if welcome, following the same pattern as the recent widget PRs (#1051, #1052, #1055-#1058).

Sketch of what we have in mind:

- `LVGLGifWidget`: `lv_gif_create(parent)` + `lv_gif_set_src(obj, source)`, with the source being the raw (undecoded) GIF file bytes.
- The main design question is the data pipeline: GIFs need the raw file bytes, unlike the current bitmap pipeline which emits decoded LVGL image descriptors. Two options: embed the raw file as a byte array in the build output (fonts-style), or reference it as a file when the build targets a file system (`imageExportMode: source`).
- Preview: the 8.4.0 editor wasm already exports `lv_gif_create`/`lv_gif_set_src`/`lv_gif_restart`, but the 9.2.2/9.4.0/9.5.0 runtimes don't seem to have `LV_USE_GIF` enabled — would you consider enabling it there so the widget can also be previewed on 9.x? (Without it the widget would show a placeholder in the editor, like the Lottie widget does today.)
- `lv_gif_restart` could later be exposed as a native action (restart animation on event).

If you would rather implement it yourself, that's fine too — a rough timeline would help us decide whether to wait or ship a fork-local widget in the meantime.

yirujikeshenshihao, 2026-09-09:
Thank you very much for your reply. According to your description, the technical link is very correct. It does involve a large amount of resources for gifs. For small gifs, I think we can do it according to the first plan you mentioned. If it is a larger gif, the method I am currently using is to read the SD card and then use a gif decoder.

IWILLTBEST, 2026-09-26:
Good news — this is now done. ✅

Since updates on the official repo have been slower lately, I've been continuing the project in an independent repository: **[IWILLTBEST/studio](https://github.com/IWILLTBEST/studio)** (many thanks to the EEZ Studio team for the original project).

GIF playback is fully working there, with LVGL 9.6.0 support:

- The full `lv_gif_*` API family is exported to the Run simulator's wasm runtime
- Verified on real hardware: **ESP32-P4 + Waveshare 7" display + IDF 5.5.1** — two GIF widgets playing at the same time (the 9.5.0 GIF rendering regression is fixed in 9.6.0)
- Exported C code is compatible with LVGL 9.5 / 9.6 output

Release notes: [v0.32.0](https://github.com/IWILLTBEST/studio/releases/tag/v0.32.0) (bilingual 中文 / English)

@yirujikeshenshihao feel free to give it a try — feedback is welcome!

---

**中文说明**：这个问题已经在我的独立仓库 [IWILLTBEST/studio](https://github.com/IWILLTBEST/studio) 中解决——支持 LVGL 9.6.0，模拟器 wasm 运行时内置完整的 `lv_gif_*` API，并在 ESP32-P4 + Waveshare 7 寸屏上实机验证了双 GIF 同时播放。详见 [v0.32.0 Release](https://github.com/IWILLTBEST/studio/releases/tag/v0.32.0)，欢迎试用反馈！

yirujikeshenshihao, 2026-09-26:
Thank you very much for your support, I will do the use test, if there is new feedback, I will have a new submission.