eez-open/studio #759: [LVGL]Can you add a QR code？

yirujikeshenshihao, 2025-03-19:
I need to display a QR code in LVGL, but I found that this control is not available. However, QR codes exist in other instrument design cases. Can this be added in the next version?

markwightman, 2025-04-01:
Hi, 

I Understand how busy you are with all the requests. just checking is the Development still on schedule to be started within the week on this update as I have donated the full amount?. Hopefully not too much to implement as LVGL already supports the QR Code, I guess the implementation will be a new flow action item?

Thank you

prasimix, 2025-04-02:
Hi Mark, thanks for the donation, the task will be added to v0.23.1 soon.

mvladic, 2025-04-02:
>  I guess the implementation will be a new flow action item?

Actually, it will be a new widget type: QRCode. Also, there will be a new LVGL action item to update QRCode data (string) during runtime if such a feature is needed.

mvladic, 2025-04-02:
Sneak peek:

[image]

markwightman, 2025-04-02:
Looks Great, Thank you. Brilliant a widget easy to drop in!

yirujikeshenshihao, 2025-04-03:
> Looks Great, Thank you. Brilliant a widget easy to drop in!

Thank you very much for Mark's donation, and also very much for the modifications made by mvladic. It seems that I can use this feature in the next version

closed 2025-04-06 by prasimix

mhaberler, 2025-04-08:
not sure if it's relevant here, anyway:

one cannot only display, but also scan and decode QRcodes with microcontrollers if the device has a camera and reasonable memory

here's an example for the M5Stack CoreS3: https://github.com/mhaberler/cores3-read-qrcode