eez-open/studio #761: [LVGL] When `Flow support` is enabled, errors occur in `ui.h` and `ui.c`

SinglWolf, 2025-03-21:
When `Flow support` is enabled, errors occur in `ui.h` and `ui.c`
In `ui.h`:
[image]
Why use this incomprehensible construction
```c
#if !defined(EEZ_FOR_LVGL)
#warning "EEZ_FOR_LVGL is not enabled"
#define EEZ_FOR_LVGL
#endif
```
when you can just add
```c
#include <eez-framework.h>
```
In `ui.c`:
Including on the header file does not work
```c
#include <eez/core/vars.h>
```
[image]
A working variant
```c
#include "ui.h"
#if defined(EEZ_FOR_LVGL)
#include <eez/core/vars.h>
#endif
```

mvladic, 2025-03-23:
If you are using [eez-framework library](https://github.com/eez-open/eez-framework), to prevent warning "EEZ_FOR_LVGL is not enabled" you need to add `-DEEZ_FOR_LVGL` to your compiler build command. But, I recommend to enable "Generate source code for EEZ flow engine (eez-framework)" option in Settings - Build (it is enabled by default when creating a new project), because in that case you don't need to link with [eez-framework library](https://github.com/eez-open/eez-framework).

[image]

SinglWolf, 2025-03-23:
> If you are using [eez-framework library](https://github.com/eez-open/eez-framework), to prevent warning "EEZ_FOR_LVGL is not enabled" you need to add `-DEEZ_FOR_LVGL` to your compiler build command.

But why should I make changes to the compilation settings of my project if the solution I have proposed allows it to work out of the box?

> I recommend to enable "Generate source code for EEZ flow engine (eez-framework)" option in Settings - Build (it is enabled by default when creating a new project), because in that case you don't need to link with [eez-framework library](https://github.com/eez-open/eez-framework).

Your recommendations certainly make sense.  In my project, I use `arduino-ESP32` as a component of the `ESP-IDF` framework. The problem is described in https://github.com/eez-open/eez-framework/issues/22
It's easier for me to set up the `eez-framework` as an `esp-idf` component once and forget than to edit  "CMakeLists.txt " in the `main` project directory and configure the `eez-framework` files via `idf_component_register`.

mvladic, 2025-03-23:
If eez-framework library is used inside LVGL project then `EEZ_FOR_LVGL` **must be defined** so this warning stays.

SinglWolf, 2025-03-23:
> If eez-framework library is used inside LVGL project then `EEZ_FOR_LVGL` **must be defined** so this warning stays.

Then need to do this:
```c
#if ARDUINO
#include <eez-framework.h>
#else
#if !defined(EEZ_FOR_LVGL)
#warning "EEZ_FOR_LVGL is not enabled"
#define EEZ_FOR_LVGL
#endif
```

mvladic, 2025-03-23:
> Your recommendations certainly make sense. In my project, I use arduino-ESP32 as a component of the ESP-IDF framework. The problem is described in https://github.com/eez-open/eez-framework/issues/22
It's easier for me to set up the eez-framework as an esp-idf component once and forget than to edit "CMakeLists.txt " in the main project directory and configure the eez-framework files via idf_component_register.

My recommendation is to use "Generate source code for EEZ flow engine (eez-framework)" and remove "eez-framework" from the esp-idf component dependencies - it is so much less hassle.

mvladic, 2025-03-25:
Can I close this issue?

SinglWolf, 2025-03-25:
> Can I close this issue?

Yes, it's not relevant yet

closed 2025-03-25 by mvladic