eez-open/studio #762: [LVGL] When `Flow support` is turned off, errors occur in `screens.c`

SinglWolf, 2025-03-23:
Due to the https://github.com/eez-open/studio/issues/761 and https://github.com/eez-open/eez-framework/issues/22 errors, I decided to disable `Flow support`.
And after clicking on the `Build` button, I was surprised to see this in the `screens.c` file:
[image]
Double `#include <string.h>`. I also saw the function call `eez_flow_get_selected_theme_index()` from `eez-framework`, which should not be there.
```c
#include <string.h>

#include "screens.h"
#include "images.h"
#include "fonts.h"
#include "actions.h"
#include "vars.h"
#include "styles.h"
#include "ui.h"

#include <string.h>

objects_t objects;
lv_obj_t *tick_value_change_obj;

void create_screen_main() {
    lv_obj_t *obj = lv_obj_create(0);
    objects.main = obj;
    lv_obj_set_pos(obj, 0, 0);
    lv_obj_set_size(obj, 320, 240);
    lv_obj_clear_flag(obj, LV_OBJ_FLAG_PRESS_LOCK|LV_OBJ_FLAG_CLICK_FOCUSABLE|LV_OBJ_FLAG_GESTURE_BUBBLE|LV_OBJ_FLAG_SNAPPABLE|LV_OBJ_FLAG_SCROLLABLE|LV_OBJ_FLAG_SCROLL_ELASTIC|LV_OBJ_FLAG_SCROLL_MOMENTUM|LV_OBJ_FLAG_SCROLL_CHAIN_HOR|LV_OBJ_FLAG_SCROLL_CHAIN_VER);
    lv_obj_set_scrollbar_mode(obj, LV_SCROLLBAR_MODE_OFF);
    lv_obj_set_scroll_dir(obj, LV_DIR_NONE);
    lv_obj_set_scroll_snap_x(obj, LV_SCROLL_SNAP_NONE);
    lv_obj_set_scroll_snap_y(obj, LV_SCROLL_SNAP_NONE);
    lv_obj_set_style_bg_color(obj, lv_color_hex(theme_colors[eez_flow_get_selected_theme_index()][1]), LV_PART_MAIN | LV_STATE_DEFAULT);
```
Dear developers, I have already discovered several critical errors that you have not responded to in any way.
I do not know if you are busy with more important things, but it is at least rude to ignore user requests.
I've been looking for a decent replacement for `SLS` for a long time and thought that `EEZ Studio`would meet my humble needs, but obviously I was wrong.

SinglWolf, 2025-03-23:
In the `screens.c` file, need to add this instead of  just `#include <string.h>`
```c
#if ARDUINO
#include <Arduino.h>
#else
#include <string.h>
#endif
```
Because in Arduino, a string variable or a string constant is defined like this:
```cpp
#include <Arduino.h>

String foo = "Hello, world!"
const String bar = "Welcome!";
```

mvladic, 2025-03-23:
We are using `#include <string.h>` for the `strcmp`.

SinglWolf, 2025-03-23:
Then so
```c
#if ARDUINO
#include <Arduino.h>
#endif

```

mvladic, 2025-03-23:
There is no need to include `Arduino.h` in screens.c. 

However, the problem you reported in this issue with the `eez_flow_get_selected_theme_index` function call is real problem and I'm currently investigating it.

SinglWolf, 2025-03-23:
> There is no need to include `Arduino.h` in screens.c.

Ok. Not relevant yet.

mvladic, 2025-03-23:
> I also saw the function call eez_flow_get_selected_theme_index() from eez-framework, which should not be there.

Fixed.

SinglWolf, 2025-03-23:
It's already the middle of the night, I'll check tomorrow, thanks.

SinglWolf, 2025-03-26:
I decided to follow your advice and enabled code generation `eez-framework`, in the project and during the compilation of the main project I began to receive this error
[image]
It turned out that the `#include "screens.h"` does not work because of the
[image]
Is this a mistake or am I doing something wrong?

mvladic, 2025-03-26:
Just add `#include "ui/screens.h"` in `console.cpp`.

SinglWolf, 2025-03-26:
> Just add `#include "ui/screens.h"` in `console.cpp`.

OMG! I have a dozen files where I need to make this correction!
Wouldn't replacing 
```c
#if !defined(EEZ_FOR_LVGL)
#include "screens.h"
#endif
``` 
 just to
 ```c
#include "screens.h"
```
 in `ui.h` for fix this problem for the entire project?

mvladic, 2025-03-26:
Yes, you can do that. Just change `ui.h` file template in your project:

[image]

closed 2025-03-27 by prasimix