eez-open/studio #766: [LVGL]AnimationImage adds the function of batch selecting images

yirujikeshenshihao, 2025-03-25:
I can achieve GIF effects by using AnimationImage, but when I have many PNG files, I can only add them one by one, which is a bit time-consuming. I hope to add the function of batch selecting images

mvladic, 2025-03-25:
Just for the record, there is already a feature to add multiple images:

[image]

Still, this can be improved furtherly for the AnimationImage widget, so it can be done in the single step.

mvladic, 2025-03-25:
You can now add multiple images at once like this:

[image]

yirujikeshenshihao, 2025-03-25:
Great! Do I need to update the software?

   

 
anim_img2.gif (view on web)

 
 
anim_img2.gif (view on web)

mvladic, 2025-03-25:
This is now in the master branch. You have two options, either you can build [master version from the source](https://github.com/eez-open/studio?tab=readme-ov-file#build-and-run-from-source-all-operating-systems) (recommended) or wait for the next version 0.23 to be released.

yirujikeshenshihao, 2025-03-26:
> 现在它位于主分支中。你有两个选择，要么[从源代码构建主版本](https://github.com/eez-open/studio?tab=readme-ov-file#build-and-run-from-source-all-operating-systems)（推荐），要么等待下一个版本0.23发布。

Thank you very much. I can use this feature in the next version

closed 2025-03-26 by yirujikeshenshihao