eez-open/studio #771: [LVGL] Generated eez-flow.cpp contains too many arguments in lv_dropdown_set_selected()

vincurekf, 2025-03-31:
**Describe the bug**
When I copy the generated files over to my lvgl project, I always have to delete `LV_ANIM_OFF` arguments from two `lv_dropdown_set_selected()` calls, located on `4018` and `4486` lines of the generated `eez-flow.cpp`

**To Reproduce**
Steps to reproduce the behavior:
1. Create new EEZ Flow project
2. Click "Build" in EEZ GUI 
3. Copy generated codes to lvgl project
5. See lines `4018` and `4486` of `eez-flow.cpp`

**Expected behavior**
The generated `eez-flow.cpp` should not contain extra arguments in the `lv_dropdown_set_selected()` function calls.

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Version EndeavourOS

**LVGL version**
- 9.2.0

**Additional context**
I am using the `lv_port_pc_vscode` base project generated with LVGL Project Creator.
I had to add `eez-flow.cpp` to `CMakeLists.txt`.

[image]

For now I can manually delete these, but maybe thing changed in LVGL v9.2.0 and this argument should be removed when building the `eez-flow.cpp` code for LVGL version above 9.x?

vincurekf, 2025-03-31:
My mistake! I did not pull the latest LVGL changes (which was about ~800 commits) after creating the project with LVGL Project Creator. It now builds properly without any changes made to the generated `eez-flow.cpp`.

I am sorry for any stress I might caused.

closed 2025-03-31 by vincurekf

heyitsyang, 2025-07-30:
@vincurekf,  sorry to bother you, but I am encountering the same problem with EEZ Studio v0.23.2 and LVGL 9.3.0.   How did you resolve this?