eez-open/studio #794: Changing the screen from .ino

ManzaLeonardo, 2025-04-30:
Hi, I'm Leonardo. This is my first time writing here.
I'm new to this topic and need help with one thing. My project is about a round gc9a01 display with an ESP32 S3 using LVGL and Eez. I'm trying to switch displays (designed in eez studio) using a button connected to the ESP32. The problem is that I've already tried LVGL display switching commands, but they don't work. I've also tried running an action (containing an LVGL display switching action) from the .ino file, but it doesn't work either.
However, if I assign the same action to a button and press it in run mode within eez, it works.
Is there anything I should keep in mind?
Thank you very much.

mvladic, 2025-05-01:
Do you call `ui_tick`?
https://github.com/eez-open/studio/wiki/FAQ#how-to-integrate-with-my-existing-cc-project

ManzaLeonardo, 2025-05-01:
Yes, I did. this is my code.

#include "lv_conf.h"
#include <lvgl.h>
#include <TFT_eSPI.h>
#include <ui.h>           // Archivo generado por EEZ Studio
#include "esp_timer.h"
#include "vars.h"
#include "actions.h"
#include "eez-flow.h"
#include "structs.h"
#include "screens.h"

/* Configuración de la pantalla GC9A01 */
static const uint16_t screenWidth  = 240;
static const uint16_t screenHeight = 240;

static lv_disp_draw_buf_t draw_buf;
static lv_color_t buf[screenWidth * 10];

TFT_eSPI tft = TFT_eSPI(); // Instancia de la pantalla

/* Función para actualizar la pantalla */
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p) {
    uint32_t w = (area->x2 - area->x1 + 1);
    uint32_t h = (area->y2 - area->y1 + 1);

    tft.startWrite();
    tft.setAddrWindow(area->x1, area->y1, w, h);
    tft.pushColors((uint16_t *)&color_p->full, w * h, true);
    tft.endWrite();

    lv_disp_flush_ready(disp);
}

void tick_increment(void* arg) {
    lv_tick_inc(1);  // Incrementa el contador de LVGL en 1 ms
}

// Definiciones de entradas digitales.
const int switch_01 = 4;  // Pin donde está conectado el switch
const int switch_02 = 5;  // Pin donde está conectado el switch
const int switch_03 = 6;  // Pin donde está conectado el switch
const int switch_04 = 7;  // Pin donde está conectado el switch
const int switch_05 = 8;  // Pin donde está conectado el switch
const int switch_06 = 9;  // Pin donde está conectado el switch

// Definiciones de variables internas.

int32_t switch_01_te;
bool switch_01_aux = 0;
bool switch_02_aux = 0;

int32_t get_var_switch_01_te() {
    return switch_01_te;
}
void set_var_switch_01_te(int32_t value) {
    switch_01_te = value;
}

bool switch_03_state;

bool get_var_switch_03_state() {
    return switch_03_state;
}
void set_var_switch_03_state(bool value) {
    switch_03_state = value;
}

void action_cam_pan(lv_event_t *e) {
    // TODO: Implement action cam_pan here
}

void setup() {
    Serial.begin(115200);
    Serial.println("Iniciando...");

    // Configuracion de entradas digitales.
    pinMode(switch_01, INPUT_PULLUP);  // Habilita la resistencia de pull-up interna
    pinMode(switch_02, INPUT_PULLUP);  // Habilita la resistencia de pull-up interna
    pinMode(switch_03, INPUT_PULLUP);  // Habilita la resistencia de pull-up interna
    pinMode(switch_04, INPUT_PULLUP);  // Habilita la resistencia de pull-up interna
    pinMode(switch_05, INPUT_PULLUP);  // Habilita la resistencia de pull-up interna
    pinMode(switch_06, INPUT_PULLUP);  // Habilita la resistencia de pull-up interna

    /* Inicializar LVGL */
    lv_init();

    /* Inicializar la pantalla */
    tft.begin();
    tft.setRotation(2); // Ajustar orientación si es necesario

    /* Inicializar el buffer de dibujo */
    lv_disp_draw_buf_init(&draw_buf, buf, NULL, screenWidth * 10);

    /* Configurar el controlador de la pantalla */
    static lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.hor_res = screenWidth;
    disp_drv.ver_res = screenHeight;
    disp_drv.flush_cb = my_disp_flush;
    disp_drv.draw_buf = &draw_buf;
    lv_disp_drv_register(&disp_drv);

    /* Inicializar la UI generada por EEZ Studio */
    ui_init();

// Configura un temporizador de hardware para llamar a tick_increment cada 1 ms
    esp_timer_create_args_t timer_args = {
        .callback = &tick_increment,
        .arg = NULL,
        .dispatch_method = ESP_TIMER_TASK,
        .name = "lv_tick_timer"
    };
    esp_timer_handle_t timer;
    esp_timer_create(&timer_args, &timer);
    esp_timer_start_periodic(timer, 1000);  // 1000 µs = 1 ms
    Serial.println("Configuración completada");

}

void loop() {
    lv_timer_handler(); // Permitir que LVGL actualice la pantalla
    delay(5);
    ui_tick(); // Actualiza la UI de EEZ Studio

    int switchState_01 = digitalRead(switch_01);  // Lee el estado del switch_01
    int switchState_02 = digitalRead(switch_02);  // Lee el estado del switch_02
        switch_03_state = digitalRead(switch_03);  // Lee el estado del switch_02

  if ((switchState_01 == LOW) && (switch_01_aux == LOW)) {
 switch_01_te ++;
 switch_01_aux = 1;
  } 
  if ((switchState_02 == LOW) && (switch_02_aux == LOW)) {
 switch_01_te --;
 switch_02_aux = 1;

   } 

if (switchState_01 == HIGH) switch_01_aux = 0;
if (switchState_02 == HIGH) switch_02_aux = 0;
if (switch_03_state == LOW){
}
  Serial.println("Botón presionado, ejecutando acción...");  
  }
}

And this is the action:

[image]

ManzaLeonardo, 2025-05-09:
Anyone can help?
Run a user action from .ino script will work too, but i don't find the way to run it.