eez-open/studio #802: Performance issue while use large font file

aleccy, 2025-05-17:
**Describe the bug**
When select a large font file, such as Chinese font file (as below image). The studio will be very slow while loading a poject file or copy-paste(may need to wait for more than 5 minutes).

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Add a large font file, you may get it from https://fontmeme.com/fonts/alibaba-puhuiti-font/
2. Set font size to 16 and ranges 0-99999
3. Create any object with the added font.
4. Very slow while build project or copy-paste or load project

**Environment and version**
 - OS: Windows
 - Version 0.23.2

**Project type
- LVGL
** LVGL version (if used)**
- 9.x

rayene01, 2025-11-20:
I get the same issues here. I hope they fix that soon.

mvladic, 2026-02-01:
The reason for this is when loading project we load all the font glyphs and when font is selected it shows those glyphs automatically. This is now optimized, glyphs are not loaded by default and to view them you need to press View Glyphs button:

[image]

aleccy, 2026-02-04:
Thanks a lot for the fix. But this fix takes in another issue, when a design uses several fonts, how can I load these fonts together with one button, currently should I click view glyphs on every font?