eez-open/studio #803: How to use the global array of strings?

HarshMarsh, 2025-05-17:
I am working on a project where I design my UI in EEZ studio and export the files to Arduino IDE to then run my project. I have a need to make an array of strings whose components I intend to show as options in a dropdown list. I made a variable of type array:string, but can’t seem to find any examples on how to interface with the variable from the vars.h/vars.cpp file. Can anyone advise how to get string data from my main code into this variable? 

Here is what is generated in my vars.h file after creating this variable: (named "available_networks")

[image]

JanBesta, 2025-05-17:
Hey, unfortunately I have no solution for you, but I am here, because I have the same problem - asked on discord, but didn't recieved any answer, and yes, I couldn't find any examples how to read and write data to arrays neither. GPT adviced me some set And get functions but I doubt it works - couldn't get IT working in my project, still compile error. So I need to understand this too

arturv2000, 2025-05-21:
Hello

Are you using only `LVGL` or `LVGL with EEZ Flow`?

In the case of using just `LVGL` you should have declaration for the `set` and `get` for that variable created:

```c
// Flow global variables

enum FlowGlobalVariables {
    FLOW_GLOBAL_VARIABLE_VAR_TEST_STRING = 0
};

// Native global variables

extern const char *get_var_var_test_string();
extern void set_var_var_test_string(const char *value);
```

Note the `get` and `set` functions declaration. It is necessary to implement those functions.

In LVGL, to pass information for a dropdown, the string must be separated by `\n`, something like this:

```c
static char var_test_string[256] = {"Op1\nOp2\nOp3\nOp4\nOp5\nOp6\nOp7\nOp8\nOp9\nOp10"};

const char *get_var_var_test_string(){
    return (const char *)var_test_string;
}
```

It is not really an "array" of strings, each line to show on dropdown is separated by character '\n'. You can create the string using `strcpy, `snprintd` whatever is more useful for you, in case of Arduino you can leverate the `string` variable type, and just return `myString.c_str()` in the function.

[image]

JanBesta, 2025-05-21:
Firstly thanks for answer, but I am using LVGL with EEZ Flow... Does it work the same? I am doing all the logic of setting correct string from dropdown to arrays of strings using index etc. and all of that works in simulation. What I need is to create JSON with ArduinoJSON library and send data by UART. Tomorrow I can edit this message with the action where is base logic but it doesn't compile.. Could you please try to write me a short variable set to some other variable?

BES0057, 2025-06-04:
I have good news for you I got it working....

I will place here exactly my case:
the variable (array of  strings): FLOW_GLOBAL_VARIABLE_INGREDIENTS_ARRAY = 6,

Code where I print each element of that array to serial monitor to see if I read the correctly:

#define INGREDIENTS_ARRAY_ID 6

using namespace eez;

void printIngredientsArray()
{
  Value value = flow::getGlobalVariable(INGREDIENTS_ARRAY_ID);

  if (!value.isArray()) {
      Serial.println("Ingredients is not an array.");
      return;
  }

  ArrayValue *array = value.getArray();
  if (!array) {
      Serial.println("Array pointer is null!");
      return;
  }

  for (uint32_t i = 0; i < array->arraySize; i++) {
      const Value &item = array->values[i];

      if (item.isString()) {
          Serial.print("Ingredient[");
          Serial.print(i);
          Serial.print("]: ");
          Serial.println(item.getString());
      } else {
          Serial.print("Ingredient[");
          Serial.print(i);
          Serial.println("]: (not a string)");
      }
  }
}

and I call this function in loop() and it works really good if any element of that array is changed in display I can immediately see changes. I will continue with array of integers, then with setvariable by native code and keep you updated if you want - answer please, otherwise I will not post other functions.

links which I used to get information and sent it to AI:
https://github.com/eez-open/native-interface-lvgl-with-flow/tree/master?tab=readme-ov-file
https://github.com/eez-open/studio/issues/260

JanBesta, 2025-06-04:
I have good news for you I got it working....

I will place here exactly my case:
the variable (array of  strings): FLOW_GLOBAL_VARIABLE_INGREDIENTS_ARRAY = 6,

Code where I print each element of that array to serial monitor to see if I read the correctly:

```
#define INGREDIENTS_ARRAY_ID 6

using namespace eez;

void printIngredientsArray()
{
  Value value = flow::getGlobalVariable(INGREDIENTS_ARRAY_ID);

  if (!value.isArray()) {
      Serial.println("Ingredients is not an array.");
      return;
  }

  ArrayValue *array = value.getArray();
  if (!array) {
      Serial.println("Array pointer is null!");
      return;
  }

  for (uint32_t i = 0; i < array->arraySize; i++) {
      const Value &item = array->values[i];

      if (item.isString()) {
          Serial.print("Ingredient[");
          Serial.print(i);
          Serial.print("]: ");
          Serial.println(item.getString());
      } else {
          Serial.print("Ingredient[");
          Serial.print(i);
          Serial.println("]: (not a string)");
      }
  }
}
```

and I call this function in loop() and it works really good if any element of that array is changed in display I can immediately see changes. I will continue with array of integers, then with setvariable by native code and keep you updated if you want - answer please, otherwise I will not post other functions.

links which I used to get information and sent it to AI:
https://github.com/eez-open/native-interface-lvgl-with-flow/tree/master?tab=readme-ov-file
https://github.com/eez-open/studio/issues/260

JanBesta, 2025-06-06:
Glad I can help :) , because I know exactly how frustrating it is. I wonder why there is no such a step-by-step manual with exact example because I know that many people are trying to find this... I will place here getting arrays of integers (works the same but instead of "getString" and "isString" it is neccesary to write "getInt32" and "isInt32" because there is no way to set size of elements inside of arrays of integers - default of eez studio is int32). I will place here also native bool after I will achiev setting / modifying elements from native code

Good luck

SinglWolf, 2025-07-23:
> I will try to answer any question and I hope that everybody who tries to find this will find it. I am thinking of placing the exact same message with code to discord.

@JanBesta, I don't have access to discord. Please post all the information in https://github.com/eez-open/studio/discussions
Thank you.