eez-open/studio #810: Has anyone succeeded with LVGL 9.x?

nishad2m8, 2025-06-16:
Hi,
Has anyone tried and successfully used LVGL version 9.x? I tested it on two boards with different display flushing methods. The raw LVGL "Hello World" and UI exported from SquareLine Studio are working fine. But the UI exported from EEZ Studio is not working.

Any advice or help would be appreciated!

txf-, 2025-06-18:
Just to get more info, what does "not working" mean exactly? Does not compile? Does not display?

nishad2m8, 2025-06-18:
It's compiled and flashed without any errors. display shows nothing or wrong color. 
See the attached generated file and screenshot.

[ui.zip](https://github.com/user-attachments/files/20799534/ui.zip)

[image]

brightproject, 2025-06-22:
> It's compiled and flashed without any errors. display shows nothing or wrong color.

`Lvgl 9.3` has a different way of creating a display, perhaps you should remove something like this from your initialization:

> lv_disp_t * dispp = lv_disp_get_default();
> lv_theme_t * theme = lv_theme_default_init(dispp, lv_palette_main(LV_PALETTE_BLUE), lv_palette_main(LV_PALETTE_RED), false, LV_FONT_DEFAULT);
> lv_disp_set_theme(dispp, theme);

Show your file code `ui.c`

txf-, 2025-06-23:
I should note that I have had no issues with lvgl 9.2 and exporting a project with version 9.x, The only change I had to make was to change the include paths from <lvgl/lvgl.h> to "lvgl.h". 

I'm using arduino and bodmers etft_spi library.

nishad2m8, 2025-06-25:
@txf-  I just tried it on the M5Stack Dial, and it's working. I think it works with TFT_eSPI  based libraries. 

Earlier, I had tried it on an AMOLED display and an e-paper display, but it didn’t work. However, without changing anything, the UI exported from other apps is working.