eez-open/studio #812: lv_dropdown_set_selected with LVGL 9.3

atlaste, 2025-06-20:
**Describe the bug**

eez-flow.cpp (and some other files) that get generated include this code:

```c++
 #if LVGL_VERSION_MAJOR >= 9 && LVGL_VERSION_MINOR >= 3
                        lv_dropdown_set_selected(target, intValue, LV_ANIM_OFF);
#else
                        lv_dropdown_set_selected(target, intValue);
#endif
```

Unfortunately, LVGL 9.3 uses this signature:

```c++
lv_dropdown_set_selected(target, intValue);
```

In other words, you get a nasty compiler error every time you re-generate.

**Desktop (please complete the following information):**

 - OS: Windows, ESP32, the lot...

** LVGL version (if used)**

- 9.3

frankalicious, 2025-07-24:
I added a [pull request](https://github.com/eez-open/eez-framework/pull/28) in eez-framework for this issue.

mvladic, 2025-09-04:
Fixed in 0.24.0

closed 2025-09-04 by mvladic