eez-open/studio #841: Cannot customize tab headers

dmitrij9992905, 2025-08-28:
Hello everyone!

I'm trying to customize tab view, and I cannot find any parameters what is in charge of tab headers style
How can I customize them?

For example, make them transparent

[image]

dmitrij9992905, 2025-08-28:
I use EEZStudio v0.23.2
I tried to find the property, but it has effect just on tab contains and not on tab header
[image]

mvladic, 2025-08-29:
Check the Styled Tabview example:

[image]

Just add two Container widgets as children to the Tabivew. These container widgets will be for the Bar and Content part. Inside Content you can add Tab widgets:

[image]

Then you can modify styles of the Bar widget.

mvladic, 2026-09-06:
Fixed. Customization of tab headers now works in LVGL version 9.x the same way it works in version 8.x, see [the following comment](https://github.com/eez-open/studio/issues/841#issuecomment-3236793009).

thisisrandy, 2026-09-06:
Talk about [a quick response](https://github.com/eez-open/studio/issues/1007)! However, I'm afraid there's still an issue. I just cloned and built the repo per the readme instructions on Ubuntu 26.04. At first glance, the Styled Tabview example works perfectly when targeting LVGL 9.x. However, when I open my project, which uses a project style to control the "Items part -> CHECKED" styling, changes to the project style are *not* reflected in the widget. Specifically, when I look at the widget's styling, changes to the project style appear in the value box, but they must be manually enabled instead of being automatically reflected the way I understand project styles to work. I tried randomly editing some other project style in the same project (button styling, in this case), and my changes were immediately reflected on the widgets using it, so there's a clear discrepancy.

mvladic, 2026-09-06:
Thanks for reporting that issue too! It should be fixed now.

thisisrandy, 2026-09-07:
Confirmed fixed on both counts. Obviously I can just build the repo for now, but when can we expect to see this in a release? Thanks very much for your quick work 😀.

mvladic, 2026-09-07:
We are planning to release it in a week or two. There are some PR's waiting to be reviewed and merged and after that we will release it.